python-control/python-control #768: pole and zeros precisions

RafaelChavesPB, 2022-09-08:
the control.zeros() and control.pole() functions are returning 0 values as non-zeros values 

2.00000000e+00+0.j -> 2.0
-4.71844785e-16+1.j -> 1j
-4.71844785e-16-1.j -> -1j

ilayn, 2022-09-08:
They are as far as floating point numbers are considered, zeros.

murrayrm, 2022-09-09:
@RafaelChavesPB Can you post the example that generated these results?  My guess is that this is just do to the underlying numerical values generated by solving for the roots of the denominator polynomial (so not something in the `control` package per se).

roryyorke, 2022-11-06:
FWIW, this gives the same result as posted:

```
In [16]: ct.poles(ct.tf(1,[1,2]) * ct.tf(1,[1,0,1]))
Out[16]: array([-2.00000000e+00+0.j,  4.71844785e-16+1.j,  4.71844785e-16-1.j])
```

which is obviously due to round-off, as @ilayn pointed out.

This answer isn't *quite* the same as given by `np.roots`, but I don't believe the difference matters:

```
nI [30]: np.roots((ct.tf(1,[1,2]) * ct.tf(1,[1,0,1])).den[0][0])
Out[30]: array([-2.00000000e+00+0.j,  4.16333634e-16+1.j,  4.16333634e-16-1.j])
```

The result can be made prettier with `np.printoptions` (or `np.set_printoptions`):

```
In [31]: with np.printoptions(suppress=True): print(ct.poles(ct.tf(1,[1,2]) * ct.tf(1,[1,0,1])))
[-2.+0.j  0.+1.j  0.-1.j]
```

closed 2022-11-20 by murrayrm