python-control/python-control #772: Simplifying transfer functions?

amosnier, 2022-09-18:
I have come across a simple example that puzzles me a little.

```python
import control

s = control.TransferFunction.s

G1 = 1 / (1 + 3 * s)**2
G2 = 3 * (1 + 1 / 2 / s)
print(-G2 * control.feedback(G1, G2))
```
```
       -3 s^2 - 1.5 s
-----------------------------
9 s^4 + 6 s^3 + 4 s^2 + 1.5 s
```

The absence of automatic factorization and simplification here does not surprise me, but I would expect a function in the toolbox that would perform such an operation. Perhaps a `control.simplify`, which applied to the transfer function above would return the following function.

```
       -3 s - 1.5
-----------------------------
9 s^3 + 6 s^2 + 4 s + 1.5
```
Maybe I don't know where to look, but I have not found such a function.

For the record, sympy allows me to do:

```python
import sympy as sp

s = sp.symbols('s')

G1 = 1 / (1 + 3 * s)**2
G2 = 3 * (1 + 1 / 2 / s)

def feedback(G1, G2, sign=-1):
    return G1 / (1 - sign * G1 * G2)

G = -G2 * feedback(G1, G2)

print(G)
print(sp.simplify(G))
print(sp.expand(sp.simplify(G)))
print(sp.simplify(sp.expand(sp.simplify(G))))
```

```
(-3 - 1.5/s)/((3*s + 1)**2*((3 + 1.5/s)/(3*s + 1)**2 + 1))
(-3*s - 1.5)/(s*(3*s + 1)**2 + 3*s + 1.5)
-3*s/(9*s**3 + 6*s**2 + 4*s + 1.5) - 1.5/(9*s**3 + 6*s**2 + 4*s + 1.5)
(-3*s - 1.5)/(9*s**3 + 6*s**2 + 4*s + 1.5)
```

ilayn, 2022-09-18:
This sounds very simple for integer coefficients but gets very tricky very quickly.Especially factoring the polynomials is an unstable operation and you can't rely on it too much. In fact this is an active research area. 

Here is one implementation of LCM of polynomials https://github.com/ilayn/harold/blob/3fec03d551a8668df1c27c7a0aa8a032ee6c5659/harold/_polynomial_ops.py#L12

I do this in `harold` and it costed me a lot of time to get it right. Even now I can see a few places where I need to do better with regards to controllability subspace computation

amosnier, 2022-09-18:
@ilayn, thanks a lot for your answer.

> This sounds very simple for integer coefficients

Integer coefficients as opposed to what? When does it start to be tricky?

I did not know about `harold`. Does it have all the functionality Python Control Systems has?

ilayn, 2022-09-18:
> Integer coefficients as opposed to what? 

Floats. Read more about it on [Wilkinson's polynomial](https://en.wikipedia.org/wiki/Wilkinson%27s_polynomial)

juanodecc, 2022-09-20:
Try minreal

import control

s = control.TransferFunction.s

G1 = 1 / (1 + 3 * s)**2
G2 = 3 * (1 + 1 / 2 / s)
print(control.minreal(-G2 * control.feedback(G1, G2)))

amosnier, 2022-09-24:
@juanodecc, thanks a lot, I had missed the text

> cancelling pole-zero pairs in transfer functions

in the documentation of that function. Gives me:

```
         -0.3333 s - 0.1667
------------------------------------
s^3 + 0.6667 s^2 + 0.4444 s + 0.1667

```
which is quite good.

closed 2022-09-24 by amosnier