python-control/python-control #778: Problems by installing on Linux without the "setup.py" file

robertobucher, 2022-10-11:
The "setup.py" file is disappeared in python-control and slycot. 

Basically it is possible to install the files in debian testing using the command "pip install python-control" and "pip install slycot", but in debian stable seems that the provided slycot version is not compatible with the python3-numpy version of this release.

I'll tried to get the git version of python-control and I've installed it using the command "pip install ."  as described in the README.md file and the result is

Successfully installed UNKNOWN-0.9.3.dev74+g61ef12b

Thanks fot hints

bnavigator, 2022-10-11:
Yes, setuptools style `python setup.py install` is deprecated. We moved to PEP517 style.

The package name value `UNKNOWN` is a bit suspicious. Did you `git clone` python-control and run `pip install .` from the toplevel directory?

`pip install slycot` will try to compile the Slycot sources. Note that you apparently try to mix system packages with PyPI packages. In this case you need to install the numpy headers from Ubuntu (some `dev` suffix package) and a Fortran compiler. And make sure that pip does not pull its own numpy wheel into the same environment.

If that is not your intention, make sure to install everything from PyPI into system-indenpendet venv. Then the Ubuntu version should not matter.

robertobucher, 2022-10-11:
Usually I've still installed Slycot and built the code (no problems under Linux). The problem appears only with debian stable and not debian testing. Probably it will be corrected using the numpy-dev library.

For python-control:

$ git clone https://github.com/python-control/python-control.git
$ cd python-control
$ sudo pip install --dry-run .
Processing /home/bucher/python-control
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Would install UNKNOWN-0.9.3.dev74+g61ef12b
 
The flag --dry-run is only to show the result.

This the result with the -v flag

$ sudo pip install --dry-run .
Processing /home/bucher/python-control
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Would install UNKNOWN-0.9.3.dev74+g61ef12b

bucher@debian:~/python-control$ sudo pip install -v --dry-run .
Using pip 22.2 from /usr/lib/python3/dist-packages/pip (python 3.10)
Processing /home/bucher/python-control
  Running command pip subprocess to install build dependencies
  Collecting setuptools
    Using cached setuptools-65.4.1-py3-none-any.whl (1.2 MB)
  Collecting setuptools-scm
    Using cached setuptools_scm-7.0.5-py3-none-any.whl (42 kB)
  Collecting wheel
    Using cached wheel-0.37.1-py2.py3-none-any.whl (35 kB)
  Collecting tomli>=1.0.0
    Using cached tomli-2.0.1-py3-none-any.whl (12 kB)
  Collecting packaging>=20.0
    Using cached packaging-21.3-py3-none-any.whl (40 kB)
  Collecting typing-extensions
    Using cached typing_extensions-4.4.0-py3-none-any.whl (26 kB)
  Collecting pyparsing!=3.0.5,>=2.0.2
    Using cached pyparsing-3.0.9-py3-none-any.whl (98 kB)
  Installing collected packages: wheel, typing-extensions, tomli, setuptools, pyparsing, packaging, setuptools-scm
  Successfully installed packaging-21.3 pyparsing-3.0.9 setuptools-65.4.1 setuptools-scm-7.0.5 tomli-2.0.1 typing-extensions-4.4.0 wheel-0.37.1
  WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
  Installing build dependencies ... done
  Running command Getting requirements to build wheel
  running egg_info
  warning: no files found matching 'tests/*.py'
  writing manifest file 'UNKNOWN.egg-info/SOURCES.txt'
  Getting requirements to build wheel ... done
  Running command Preparing metadata (pyproject.toml)
  running dist_info
  creating /tmp/pip-modern-metadata-2q71kykd/UNKNOWN.egg-info
  writing manifest file '/tmp/pip-modern-metadata-2q71kykd/UNKNOWN.egg-info/SOURCES.txt'
  warning: no files found matching 'tests/*.py'
  writing manifest file '/tmp/pip-modern-metadata-2q71kykd/UNKNOWN.egg-info/SOURCES.txt'
  Preparing metadata (pyproject.toml) ... done
Would install UNKNOWN-0.9.3.dev74+g61ef12b
bucher@debian:~/python-control$

bnavigator, 2022-10-11:
Could you add a verbose flag, please?

```
[ben@skylab:~/src/python-control]% python3 -m venv testinstall --system-site-packages                                                                                                                                                          [0]
[ben@skylab:~/src/python-control]% . testinstall/bin/activate                                                                                                                                                                                  [0]
(testinstall) [ben@skylab:~/src/python-control]% pip install -v --dry-run .                                                                                                                                                                    [0]
Using pip 22.2.2 from /home/ben/src/python-control/testinstall/lib64/python3.10/site-packages/pip (python 3.10)
Processing /home/ben/src/python-control
  Running command pip subprocess to install build dependencies
  Collecting setuptools
    Using cached setuptools-65.4.1-py3-none-any.whl (1.2 MB)
  Collecting setuptools-scm
    Using cached setuptools_scm-7.0.5-py3-none-any.whl (42 kB)
  Collecting wheel
    Using cached wheel-0.37.1-py2.py3-none-any.whl (35 kB)
  Collecting tomli>=1.0.0
    Using cached tomli-2.0.1-py3-none-any.whl (12 kB)
  Collecting packaging>=20.0
    Using cached packaging-21.3-py3-none-any.whl (40 kB)
  Collecting typing-extensions
    Using cached typing_extensions-4.4.0-py3-none-any.whl (26 kB)
  Collecting pyparsing!=3.0.5,>=2.0.2
    Using cached pyparsing-3.0.9-py3-none-any.whl (98 kB)
  Installing collected packages: wheel, typing-extensions, tomli, setuptools, pyparsing, packaging, setuptools-scm
  ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
  rpmlint 2.3.0 requires python-magic, which is not installed.
  distributed 2022.9.0 requires tornado<6.2,>=6.0.3, but you have tornado 6.2 which is incompatible.
  Successfully installed packaging-21.3 pyparsing-3.0.9 setuptools-65.4.1 setuptools-scm-7.0.5 tomli-2.0.1 typing-extensions-4.4.0 wheel-0.37.1
  Installing build dependencies ... done
  Running command Getting requirements to build wheel
  /usr/lib/python3.10/site-packages/setuptools/config/pyprojecttoml.py:104: _BetaConfiguration: Support for `[tool.setuptools]` in `pyproject.toml` is still *beta*.
    warnings.warn(msg, _BetaConfiguration)
  For maximum compatibility please make sure to include a `scheme` prefix in your URL (e.g. 'http://'). Given value: https//python-control.org
  running egg_info
  writing control.egg-info/PKG-INFO
  writing dependency_links to control.egg-info/dependency_links.txt
  writing requirements to control.egg-info/requires.txt
  writing top-level names to control.egg-info/top_level.txt
  reading manifest template 'MANIFEST.in'
  warning: no files found matching 'tests/*.py'
  adding license file 'LICENSE'
  writing manifest file 'control.egg-info/SOURCES.txt'
  Getting requirements to build wheel ... done
  Running command Preparing metadata (pyproject.toml)
  /usr/lib/python3.10/site-packages/setuptools/config/pyprojecttoml.py:104: _BetaConfiguration: Support for `[tool.setuptools]` in `pyproject.toml` is still *beta*.
    warnings.warn(msg, _BetaConfiguration)
  For maximum compatibility please make sure to include a `scheme` prefix in your URL (e.g. 'http://'). Given value: https//python-control.org
  running dist_info
  creating /tmp/pip-modern-metadata-0x3o4ap2/control.egg-info
  writing /tmp/pip-modern-metadata-0x3o4ap2/control.egg-info/PKG-INFO
  writing dependency_links to /tmp/pip-modern-metadata-0x3o4ap2/control.egg-info/dependency_links.txt
  writing requirements to /tmp/pip-modern-metadata-0x3o4ap2/control.egg-info/requires.txt
  writing top-level names to /tmp/pip-modern-metadata-0x3o4ap2/control.egg-info/top_level.txt
  writing manifest file '/tmp/pip-modern-metadata-0x3o4ap2/control.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  warning: no files found matching 'tests/*.py'
  adding license file 'LICENSE'
  writing manifest file '/tmp/pip-modern-metadata-0x3o4ap2/control.egg-info/SOURCES.txt'
  creating '/tmp/pip-modern-metadata-0x3o4ap2/control-0.9.3.dev74+g61ef12b.dist-info'
  adding license file "LICENSE" (matched pattern "LICEN[CS]E*")
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: numpy in /usr/lib64/python3.10/site-packages (from control==0.9.3.dev74+g61ef12b) (1.21.6)
Requirement already satisfied: scipy>=1.3 in /usr/lib64/python3.10/site-packages (from control==0.9.3.dev74+g61ef12b) (1.8.1)
Requirement already satisfied: matplotlib in /usr/lib64/python3.10/site-packages (from control==0.9.3.dev74+g61ef12b) (3.5.3)
Requirement already satisfied: cycler>=0.10 in /usr/lib/python3.10/site-packages (from matplotlib->control==0.9.3.dev74+g61ef12b) (0.11.0)
Requirement already satisfied: fonttools>=4.22.0 in /usr/lib/python3.10/site-packages (from matplotlib->control==0.9.3.dev74+g61ef12b) (4.37.1)
Requirement already satisfied: kiwisolver>=1.0.1 in /usr/lib64/python3.10/site-packages (from matplotlib->control==0.9.3.dev74+g61ef12b) (1.4.4)
Requirement already satisfied: packaging>=20.0 in /usr/lib/python3.10/site-packages (from matplotlib->control==0.9.3.dev74+g61ef12b) (21.3)
[... cut ...]
Requirement already satisfied: pyparsing>=2.2.1 in /usr/lib/python3.10/site-packages (from matplotlib->control==0.9.3.dev74+g61ef12b) (3.0.9)
Requirement already satisfied: python-dateutil>=2.7 in /usr/lib/python3.10/site-packages (from matplotlib->control==0.9.3.dev74+g61ef12b) (2.8.2)
Requirement already satisfied: six>=1.5 in /usr/lib/python3.10/site-packages (from python-dateutil>=2.7->matplotlib->control==0.9.3.dev74+g61ef12b) (1.16.0)
Would install control-0.9.3.dev74+g61ef12b
(testinstall)
```

P.S: `sudo pip` is never a good idea. Keep system packages and pip management separate at all cost.

robertobucher, 2022-10-11:
$ sudo pip install -v --dry-run .
Using pip 22.2 from /usr/lib/python3/dist-packages/pip (python 3.10)
Processing /home/bucher/python-control
  Running command pip subprocess to install build dependencies
  Collecting setuptools
    Using cached setuptools-65.4.1-py3-none-any.whl (1.2 MB)
  Collecting setuptools-scm
    Using cached setuptools_scm-7.0.5-py3-none-any.whl (42 kB)
  Collecting wheel
    Using cached wheel-0.37.1-py2.py3-none-any.whl (35 kB)
  Collecting typing-extensions
    Using cached typing_extensions-4.4.0-py3-none-any.whl (26 kB)
  Collecting packaging>=20.0
    Using cached packaging-21.3-py3-none-any.whl (40 kB)
  Collecting tomli>=1.0.0
    Using cached tomli-2.0.1-py3-none-any.whl (12 kB)
  Collecting pyparsing!=3.0.5,>=2.0.2
    Using cached pyparsing-3.0.9-py3-none-any.whl (98 kB)
  Installing collected packages: wheel, typing-extensions, tomli, setuptools, pyparsing, packaging, setuptools-scm
  Successfully installed packaging-21.3 pyparsing-3.0.9 setuptools-65.4.1 setuptools-scm-7.0.5 tomli-2.0.1 typing-extensions-4.4.0 wheel-0.37.1
  WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
  Installing build dependencies ... done
  Running command Getting requirements to build wheel
  running egg_info
  warning: no files found matching 'tests/*.py'
  writing manifest file 'UNKNOWN.egg-info/SOURCES.txt'
  Getting requirements to build wheel ... done
  Running command Preparing metadata (pyproject.toml)
  running dist_info
  creating /tmp/pip-modern-metadata-nv7dxkzw/UNKNOWN.egg-info
  writing manifest file '/tmp/pip-modern-metadata-nv7dxkzw/UNKNOWN.egg-info/SOURCES.txt'
  warning: no files found matching 'tests/*.py'
  writing manifest file '/tmp/pip-modern-metadata-nv7dxkzw/UNKNOWN.egg-info/SOURCES.txt'
  Preparing metadata (pyproject.toml) ... done
Would install UNKNOWN-0.9.3.dev74+g61ef12b

bnavigator, 2022-10-11:
> Installing collected packages: wheel, typing-extensions, tomli, setuptools, pyparsing, packaging, setuptools-scm
...
> WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager.

What I said before. You basically destroyed your apt managed python environment.

> warning: no files found matching 'tests/.py'

That does not look like anything from python-control. Are you sure you have a clean source directory?

Regardless, I fired up a `docker:stable` container and it would install a `control-0.9.3.dev` package:

<details>
<summary>`docker run -it debian:stable`</summary>
```
[ben@skylab:~/tmp]% docker run -it debian:stable

Unable to find image 'debian:stable' locally
stable: Pulling from library/debian
225ba7910a62: Pull complete 
Digest: sha256:1b1d158efc589b1eb8858acdc95bd0ff4c5344958f834b7326662da8482b3e7d
Status: Downloaded newer image for debian:stable
root@1955cf6d491f:/# apt update
Get:1 http://deb.debian.org/debian stable InRelease [116 kB]
Get:2 http://deb.debian.org/debian-security stable-security InRelease [48.4 kB]
Get:3 http://deb.debian.org/debian stable-updates InRelease [44.1 kB]
Get:4 http://deb.debian.org/debian stable/main amd64 Packages [8184 kB]
Get:5 http://deb.debian.org/debian-security stable-security/main amd64 Packages [189 kB]
Get:6 http://deb.debian.org/debian stable-updates/main amd64 Packages [6344 B]
Fetched 8587 kB in 2s (4523 kB/s)                        
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
root@1955cf6d491f:/# apt-get install git python3                        
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  ca-certificates git-man less libbrotli1 libbsd0 libcbor0 libcurl3-gnutls libedit2 liberror-perl libexpat1 libfido2-1 libgdbm-compat4 libgdbm6 libgpm2 libldap-2.4-2 libldap-common libmd0 libmpdec3 libncursesw6 libnghttp2-14 libperl5.32
  libpsl5 libpython3-stdlib libpython3.9-minimal libpython3.9-stdlib libreadline8 librtmp1 libsasl2-2 libsasl2-modules libsasl2-modules-db libsqlite3-0 libssh2-1 libx11-6 libx11-data libxau6 libxcb1 libxdmcp6 libxext6 libxmuu1 media-types
  netbase openssh-client openssl patch perl perl-modules-5.32 publicsuffix python3-minimal python3.9 python3.9-minimal readline-common xauth
Suggested packages:
  gettext-base git-daemon-run | git-daemon-sysvinit git-doc git-el git-email git-gui gitk gitweb git-cvs git-mediawiki git-svn gdbm-l10n gpm sensible-utils libsasl2-modules-gssapi-mit | libsasl2-modules-gssapi-heimdal libsasl2-modules-ldap
  libsasl2-modules-otp libsasl2-modules-sql keychain libpam-ssh monkeysphere ssh-askpass ed diffutils-doc perl-doc libterm-readline-gnu-perl | libterm-readline-perl-perl make libtap-harness-archive-perl python3-doc python3-tk python3-venv
  python3.9-venv python3.9-doc binutils binfmt-support readline-doc
The following NEW packages will be installed:
  ca-certificates git git-man less libbrotli1 libbsd0 libcbor0 libcurl3-gnutls libedit2 liberror-perl libexpat1 libfido2-1 libgdbm-compat4 libgdbm6 libgpm2 libldap-2.4-2 libldap-common libmd0 libmpdec3 libncursesw6 libnghttp2-14 libperl5.32
  libpsl5 libpython3-stdlib libpython3.9-minimal libpython3.9-stdlib libreadline8 librtmp1 libsasl2-2 libsasl2-modules libsasl2-modules-db libsqlite3-0 libssh2-1 libx11-6 libx11-data libxau6 libxcb1 libxdmcp6 libxext6 libxmuu1 media-types
  netbase openssh-client openssl patch perl perl-modules-5.32 publicsuffix python3 python3-minimal python3.9 python3.9-minimal readline-common xauth
0 upgraded, 54 newly installed, 0 to remove and 0 not upgraded.
Need to get 26.8 MB of archives.
After this operation, 124 MB of additional disk space will be used.
Do you want to continue? [Y/n] 
Get:1 http://deb.debian.org/debian stable/main amd64 perl-modules-5.32 all 5.32.1-4+deb11u2 [2823 kB]
Get:2 http://deb.debian.org/debian stable/main amd64 libgdbm6 amd64 1.19-2 [64.9 kB]
Get:3 http://deb.debian.org/debian stable/main amd64 libgdbm-compat4 amd64 1.19-2 [44.7 kB]
Get:4 http://deb.debian.org/debian stable/main amd64 libperl5.32 amd64 5.32.1-4+deb11u2 [4106 kB]
Get:5 http://deb.debian.org/debian stable/main amd64 perl amd64 5.32.1-4+deb11u2 [293 kB]
Get:6 http://deb.debian.org/debian stable/main amd64 libpython3.9-minimal amd64 3.9.2-1 [801 kB]
Get:7 http://deb.debian.org/debian-security stable-security/main amd64 libexpat1 amd64 2.2.10-2+deb11u4 [98.0 kB]
Get:8 http://deb.debian.org/debian stable/main amd64 python3.9-minimal amd64 3.9.2-1 [1955 kB]
Get:9 http://deb.debian.org/debian stable/main amd64 python3-minimal amd64 3.9.2-3 [38.2 kB]
Get:10 http://deb.debian.org/debian stable/main amd64 media-types all 4.0.0 [30.3 kB]
Get:11 http://deb.debian.org/debian stable/main amd64 libmpdec3 amd64 2.5.1-1 [87.7 kB]
Get:12 http://deb.debian.org/debian stable/main amd64 libncursesw6 amd64 6.2+20201114-2 [132 kB]
Get:13 http://deb.debian.org/debian stable/main amd64 readline-common all 8.1-1 [73.7 kB]
Get:14 http://deb.debian.org/debian stable/main amd64 libreadline8 amd64 8.1-1 [169 kB]
Get:15 http://deb.debian.org/debian stable/main amd64 libsqlite3-0 amd64 3.34.1-3 [797 kB]
Get:16 http://deb.debian.org/debian stable/main amd64 libpython3.9-stdlib amd64 3.9.2-1 [1684 kB]
[... 889 lines cut ...]
  Using cached packaging-21.3-py3-none-any.whl (40 kB)
[... cut ...]
Collecting cycler>=0.10
  Downloading cycler-0.11.0-py3-none-any.whl (6.4 kB)
Collecting pyparsing>=2.2.1
  Using cached pyparsing-3.0.9-py3-none-any.whl (98 kB)
Collecting six>=1.5
  Downloading six-1.16.0-py2.py3-none-any.whl (11 kB)
Would install Pillow-9.2.0 contourpy-1.0.5 control-0.9.3.dev74+g61ef12b cycler-0.11.0 fonttools-4.37.4 kiwisolver-1.4.4 matplotlib-3.6.1 numpy-1.23.3 packaging-21.3 pyparsing-3.0.9 python-dateutil-2.8.2 scipy-1.9.2 six-1.16.0
root@1955cf6d491f:/python-control# 

```

</details>

robertobucher, 2022-10-11:
Thanks!
Ok. I'll try to check it on a Debian stable distro and see what happens.

bnavigator, 2022-10-11:
> > warning: no files found matching 'tests/.py'
> 
> That does not look like anything from python-control. Are you sure you have a clean source directory?

Ah there is an old line in the MANIFEST.in regarding this. I removed it in #779, but it should be unrelated. And I wonder why you get the warning, while it is not there on my side.

bnavigator, 2022-10-11:
Found it: It's a Debian thing: https://github.com/pypa/setuptools/issues/3269

bnavigator, 2022-10-11:
Until the fixed pip with https://github.com/pypa/pip/pull/11466 is released, you can work around it with `DEB_PYTHON_INSTALL_LAYOUT=deb`

Nevertheless, I still **strongly** recommend **not** to run `pip` with `sudo` and let it manipulate your systemwide `dist-packages`.

closed 2022-10-11 by bnavigator