python-control/python-control #801: Segmentation Fault on MacOS - Python 3.9

SeyedMohamadMoosavi, 2022-11-26:
Installed the package using conda in a clean conda environment on MacOS Ventura M1 Pro Chip.

`conda install -c conda-forge control slycot`

However, I get segmentation fault after importing control:
`import control`
output is:
> [1]    11822 segmentation fault  python

murrayrm, 2022-11-26:
Can you post the results of `conda info`.  I'd like to see what architecture is being used.

SeyedMohamadMoosavi, 2022-11-26:
Here is log: 

           shell level : 2
       user config file : /Users/USER/.condarc
 populated config files : /Users/USER/.condarc
          conda version : 22.9.0
    conda-build version : not installed
         python version : 3.8.15.final.0
       virtual packages : __osx=13.0.1=0
                          __unix=0=0
                          __archspec=1=arm64
       base environment : /Users/USER/miniconda3  (writable)
      conda av data dir : /Users/USER/miniconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://repo.anaconda.com/pkgs/main/osx-arm64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/osx-arm64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/USER/miniconda3/pkgs
                          /Users/USER/.conda/pkgs
       envs directories : /Users/USER/miniconda3/envs
                          /Users/USER/.conda/envs
               platform : osx-arm64
             user-agent : conda/22.9.0 requests/2.28.1 CPython/3.8.15 Darwin/22.1.0 OSX/13.0.1
                UID:GID : 501:20
             netrc file : None
           offline mode : False

murrayrm, 2022-11-26:
That all looks OK.  Can you run `conda list -c` and post.  That will show which versions of various packages are installed and where they came from.

SeyedMohamadMoosavi, 2022-11-26:
> conda-forge/noarch::anyio-3.6.2-pyhd8ed1ab_0
> conda-forge/noarch::appnope-0.1.3-pyhd8ed1ab_0
> conda-forge/osx-arm64::argon2-cffi-21.1.0-py39h5161555_0
> conda-forge/noarch::asttokens-2.1.0-pyhd8ed1ab_0
> defaults/osx-arm64::attrs-22.1.0-py39hca03da5_0
> conda-forge/noarch::babel-2.11.0-pyhd8ed1ab_0
> conda-forge/noarch::backcall-0.2.0-pyh9f0ad1d_0
> conda-forge/noarch::backports-1.0-pyhd8ed1ab_3
> conda-forge/noarch::backports.functools_lru_cache-1.6.4-pyhd8ed1ab_0
> conda-forge/noarch::beautifulsoup4-4.11.1-pyha770c72_0
> conda-forge/osx-arm64::blas-2.116-openblas
> conda-forge/osx-arm64::blas-devel-3.9.0-16_osxarm64_openblas
> conda-forge/noarch::bleach-5.0.1-pyhd8ed1ab_0
> defaults/osx-arm64::bottleneck-1.3.5-py39heec5a64_0
> defaults/osx-arm64::brotli-1.0.9-h1a28f6b_7
> defaults/osx-arm64::brotli-bin-1.0.9-h1a28f6b_7
> conda-forge/osx-arm64::brotlipy-0.7.0-py39h5161555_1001
> defaults/osx-arm64::bzip2-1.0.8-h620ffc9_4
> conda-forge/osx-arm64::ca-certificates-2022.9.24-h4653dfc_0
> conda-forge/noarch::certifi-2022.9.24-pyhd8ed1ab_0
> defaults/osx-arm64::cffi-1.15.1-py39h80987f9_2
> conda-forge/noarch::charset-normalizer-2.1.1-pyhd8ed1ab_0
> conda-forge/noarch::control-0.9.2-pyhd8ed1ab_0
> defaults/osx-arm64::cryptography-38.0.1-py39h834c97f_0
> defaults/noarch::cycler-0.11.0-pyhd3eb1b0_0
> conda-forge/noarch::decorator-5.1.1-pyhd8ed1ab_0
> conda-forge/noarch::defusedxml-0.7.1-pyhd8ed1ab_0
> conda-forge/noarch::entrypoints-0.4-pyhd8ed1ab_0
> conda-forge/noarch::executing-1.2.0-pyhd8ed1ab_0
> conda-forge/osx-arm64::fftw-3.3.10-nompi_h3046061_4
> defaults/noarch::fonttools-4.25.0-pyhd3eb1b0_0
> defaults/osx-arm64::freetype-2.12.1-h1192e45_0
> defaults/osx-arm64::giflib-5.2.1-h1a28f6b_0
> defaults/osx-arm64::gmp-6.2.1-hc377ac9_3
> defaults/osx-arm64::gmpy2-2.1.2-py39h8c48613_0
> conda-forge/noarch::idna-3.4-pyhd8ed1ab_0
> conda-forge/noarch::importlib-metadata-5.1.0-pyha770c72_0
> conda-forge/noarch::importlib_resources-5.10.0-pyhd8ed1ab_0
> defaults/noarch::iniconfig-1.1.1-pyhd3eb1b0_0
> conda-forge/osx-arm64::ipykernel-5.5.5-py39h32adebf_0
> defaults/osx-arm64::ipython-8.6.0-py39hca03da5_0
> conda-forge/noarch::ipython_genutils-0.2.0-py_1
> conda-forge/noarch::jedi-0.18.2-pyhd8ed1ab_0
> conda-forge/noarch::jinja2-3.1.2-pyhd8ed1ab_1
> defaults/osx-arm64::jpeg-9e-h1a28f6b_0
> conda-forge/noarch::json5-0.9.5-pyh9f0ad1d_0
> conda-forge/noarch::jsonschema-4.17.1-pyhd8ed1ab_0
> conda-forge/noarch::jupyter_client-7.4.7-pyhd8ed1ab_0
> defaults/osx-arm64::jupyter_core-4.11.2-py39hca03da5_0
> conda-forge/noarch::jupyter_server-1.23.3-pyhd8ed1ab_0
> conda-forge/noarch::jupyterlab-3.5.0-pyhd8ed1ab_0
> conda-forge/noarch::jupyterlab_pygments-0.2.2-pyhd8ed1ab_0
> conda-forge/noarch::jupyterlab_server-2.16.3-pyhd8ed1ab_0
> defaults/osx-arm64::kiwisolver-1.4.2-py39hc377ac9_0
> defaults/osx-arm64::lcms2-2.12-hba8e193_0
> defaults/osx-arm64::lerc-3.0-hc377ac9_0
> conda-forge/osx-arm64::libblas-3.9.0-16_osxarm64_openblas
> defaults/osx-arm64::libbrotlicommon-1.0.9-h1a28f6b_7
> defaults/osx-arm64::libbrotlidec-1.0.9-h1a28f6b_7
> defaults/osx-arm64::libbrotlienc-1.0.9-h1a28f6b_7
> conda-forge/osx-arm64::libcblas-3.9.0-16_osxarm64_openblas
> defaults/osx-arm64::libcxx-14.0.6-h848a8c0_0
> defaults/osx-arm64::libdeflate-1.8-h1a28f6b_5
> defaults/osx-arm64::libffi-3.4.2-hca03da5_6
> conda-forge/osx-arm64::libgfortran-5.0.0-11_3_0_hd922786_26
> conda-forge/osx-arm64::libgfortran5-11.3.0-hdaf2cc0_26
> conda-forge/osx-arm64::liblapack-3.9.0-16_osxarm64_openblas
> conda-forge/osx-arm64::liblapacke-3.9.0-16_osxarm64_openblas
> conda-forge/osx-arm64::libopenblas-0.3.21-openmp_hc731615_3
> defaults/osx-arm64::libpng-1.6.37-hb8d0fd4_0
[... 69 lines cut ...]
> conda-forge/noarch::soupsieve-2.3.2.post1-pyhd8ed1ab_0
> defaults/osx-arm64::sqlite-3.40.0-h7a7dc30_0
> conda-forge/noarch::stack_data-0.6.1-pyhd8ed1ab_0
> defaults/osx-arm64::sympy-1.11.1-py39hca03da5_0
> pypi/pypi::tbcontrol-0.2.1-pypi_0
> conda-forge/noarch::terminado-0.17.0-pyhd1c38e8_0
> conda-forge/noarch::tinycss2-1.2.1-pyhd8ed1ab_0
> defaults/osx-arm64::tk-8.6.12-hb8d0fd4_0
> defaults/osx-arm64::tomli-2.0.1-py39hca03da5_0
> defaults/osx-arm64::tornado-6.2-py39h1a28f6b_0
> pypi/pypi::tqdm-4.64.1-pypi_0
> conda-forge/noarch::traitlets-5.5.0-pyhd8ed1ab_0
> conda-forge/noarch::typing_extensions-4.4.0-pyha770c72_0
> defaults/noarch::tzdata-2022f-h04d1e81_0
> conda-forge/noarch::urllib3-1.26.13-pyhd8ed1ab_0
> conda-forge/noarch::wcwidth-0.2.5-pyh9f0ad1d_2
> conda-forge/noarch::webencodings-0.5.1-py_1
> conda-forge/noarch::websocket-client-1.4.2-pyhd8ed1ab_0
> defaults/noarch::wheel-0.37.1-pyhd3eb1b0_0
> defaults/osx-arm64::xz-5.2.6-h1a28f6b_0
> conda-forge/osx-arm64::zeromq-4.3.4-hbdafb3b_1
> conda-forge/noarch::zipp-3.10.0-pyhd8ed1ab_0
> conda-forge/osx-arm64::zlib-1.2.13-h03a7124_4
> defaults/osx-arm64::zstd-1.5.2-h8574219_0
>

murrayrm, 2022-11-26:
I'm guessing that the problem may be with slycot, since the control package itself is just pure python.  Can you try running `conda remove --force slycot` (should remove slycot but nothing else) and then see if you can import control?

Note: this may screw up your conda environment, so if you are using this environment for other things, you might want to create a new one first.

murrayrm, 2022-11-26:
Actually, another thing to try first: see if `import slycot` causes a segmentation fault.

SeyedMohamadMoosavi, 2022-11-26:
Yes! `import slycot` causes segmentation fault.

murrayrm, 2022-11-26:
OK, we should move this issue over to the slycot issues list so the group there can have a look.  I'll do that and post the info from here.

closed 2022-11-26 by murrayrm