python-control/python-control #815: When installled along with Pytorch, fail to import torch

wuzhin, 2022-12-16:
control==0.9.2/0.9.1
torch=13.0
when running
import pytorch
an error raised
OSError: [WinError 182] 操作系统无法运行 %1。 Error loading "D:\ProgramData\Anaconda3\envs\control\lib\site-packages\torch\lib\shm.dll" or one of its dependencies.

murrayrm, 2022-12-16:
It could be a dependency issue.  This worked OK for me on MacOS 12.6.1 using Python 3.10 installed via conda (and using conda-forge for pytorch).

How are you installing python-control (conda, pip, other?) and how are you installing pytorch?

wuzhin, 2022-12-16:
> 

I've tried control(0.9.2&0..9.1) along with Pytorch(13.0&12.0) is not work, and it works on control 0.9.0 along with ANY version of Pytorch.
I use 'conda install control==version' install it and
'conda install pytorch torchvision torchaudio pytorch-cuda=11.7 -c pytorch -c nvidia' for Pytorch 13.0
'conda install pytorch==1.12.0 torchvision==0.13.0 torchaudio==0.12.0 cudatoolkit=11.6 -c pytorch -c conda-forge' for Pytrch 12.0

murrayrm, 2022-12-16:
Can you try
```
conda install -c conda-forge control
conda instal -c conda-forge pytorch
```
and see if that works (it did for me).

murrayrm, 2022-12-16:
Also, try installing python-control without slycot and see if that works (slycot brings in various math libraries that could be causing a conflict).  To do that:
```
conda install numpy scipy matplotlib
pip install control
conda install pytorch

wuzhin, 2023-01-01:
use
pip install control
works

closed 2023-01-01 by wuzhin