python-control/python-control #825: Planned release of control-0.9.3

murrayrm, 2022-12-24:
I'm planning to do a release of python-control over the winter break since I need some of the collocation techniques from PR #712 for the class I am teaching.  If there are any other critical updates we should do before the release, post a quick note here (ideally NLT 28 Dec, which I will start working on the release).

@bnavigator @sawyerbfuller

sawyerbfuller, 2022-12-27:
Hi @murrayrm I think I have everything in that I was hoping to have for the course that will be offered March, so I am set.

murrayrm, 2022-12-31:
Working on the release now...

murrayrm, 2022-12-31:
Documenting a couple of things for prosperity:
* The release instructions have been updated to reflect the use of the OS/BLAS test matrix on the final release (just to make sure things work) and the revised setup procedure (using setuptools instead of setup.py)

murrayrm, 2022-12-31:
Version 0.9.3.post2 is now released.  Had to do a few post releases to handle issues that we hadn't encountered before:

* #833: needed to update the version check test in `__init__` to get the right version information
* #835: needed to add readme information to `pyproject.toml`

closed 2022-12-31 by murrayrm