python-control/python-control #830: Incorrect nyquist plot

Avitan01, 2022-12-31:
I'm using the control package with python 3.10 in Pycharm and Jupiter lab.
I have stumbled upon a possible issue while trying to create a Nyquist plot for a simple system
```
import control as ct

s = ct.tf('s'))
plant_1 = 1/(s*(s + 5)*(s + 7)))
ct.nyquist_plot(plant_1)

```

The result I'm getting is this: 

[image]

 While in MATLAB I'm getting different results

```
clear all;
close all;
clc;

s = tf('s');
plant_1 = 1/(s*(s + 5)*(s + 7));
nyquistplot(plant_1)

```
[image]

**Is there a limitation that might be causing this issue or am I missing something?**

bnavigator, 2022-12-31:
Look at the y-axis scale. The nyquist plot from control cuts off at `max_curve_magnitude` and scales the "infinity" loop for display within the plot boundaries.
https://python-control.readthedocs.io/en/0.9.2/generated/control.nyquist_plot.html#control.nyquist_plot

henklaak, 2023-02-02:
Behaviour that will 'surprise' many, I guess. The documentation is not explicit about it and the default value of 20.

How about not drawing the large magnitudes at all? Just clip the traces at the limit with a ≫to indicate direction?

murrayrm, 2023-02-02:
If you clip the traces, then it is harder to see the number of net encirclements. Especially if there is more than one.  For example
```
sys = ct.tf([1], [1, 3, 2]) * ct.tf([1], [1, 0, 1])
ct.nyquist_plot(sys)
```
which generates:
[image]

murrayrm, 2023-02-02:
We should definitely make sure all of this is well-documented, though.

henklaak, 2023-02-03:
I'll take a stab at this, expect PR.

PR #880 (2023-03-31): Small fixes for 0.9.4 release
This PR implements a number of small fixes for issues tagged for resolution in release 0.9.4:

* Updated installation documentation to recommend conda installation first, then pip (resolves #850)
* Added more documentation about Nyquist plot scaling (resolves #830)
* Deprecated `issys()` (resolves #876)
* Changed short package name for `control.optimal` from `opt` to `obc` consistently throughout documentation.
* Fixed sphinx processing of version to allow proper links to source code (resolves #837).
* Added numpy 1.24.2 to list of exceptions for flatness test on  [... cut]
closed 2023-03-31 by murrayrm