python-control/python-control #839: control.damp(sys)

Nabil-Bishtawi, 2023-01-01:
control.damp(sys) do not deal correctly with overdamped modes and built in print function inside it should be deactivated because even if you put it wn[i],damping[i],eig[i]=control.damp(sys[i]) it is still printing every thing

roryyorke, 2023-01-08:
It's not clear what you mean by "do[es] not deal correctly with overdamped modes", though I've taken a guess (see below).  If I'm wrong, could you give us more detail, preferably as code, with inputs and actual and expected outputs?

You can prevent printing [by setting argument `doprint` to `False` ](https://python-control.readthedocs.io/en/latest/generated/control.damp.html#control.damp).

My guess is that you want `control.damp` to handle second-order systems with damping factor > 1 as second-order systems, but it instead treats these as independent first-order poles.  I think this is fine, and arguably correct: we can't know if in general if these two poles are conceptually from two different first-order physical models, or from a single over-damped second order system.

Nabil-Bishtawi, 2023-01-14:
We can determine it, assuming we have those eigenvalues (poles location):

[image]

We can follow the equations here:
[image]
1- use 14.10.3 to calculate undamped frequency.
2-use 14.10.4 to calculate damping.
3- use 14.10.2 and 14.10.5 to calculate damped frequency, and if they yield the same damped frequency, that means those two eigenvalues correspond to the same mode; if not, we choose another eigenvalue and do the steps again.

The results will be like this:
[image]

roryyorke, 2023-01-15:
You didn't say what $\lambda_n$ and $\lambda_r$ are, I've assumed they're the real pole positions.

(14.10.3) and  (14.10.4) look OK, and I think (14.10.2) is defining $\omega_d$ to be similar to the under-damped case (where it would be $\sqrt{1-\zeta^2}$.  

It's not so obvious what (14.10.1) (and (14.10.5), which can be derived from (14.10.1)) is for.

What are either over-damped $\omega_d$ values supposed to tell us?  For the case with no zeros, the response is either approximately that of a double pole, or dominated by the slow pole.  We can use the $\lambda$ values for that.

If I apply these equations to the real poles in your example, I *don't* get the values for the two definitions of $\omega_d$ to match (perhaps I made an error?).

```python
def overdamped(ln, lr):
    if not ln.imag == 0 and lr.imag == 0:
        raise ValueError('poles must be real')
    if not lr < ln < 0:
        raise ValueError('require lr<ln<0')
    
    wn = (ln * lr) ** 0.5
    zeta = -(ln + lr) / 2 / wn

    wnd1 = wn * (zeta**2 - 1)
    wnd2 = 0.5 * (ln - lr)
    print(f'{wn=:.4f}, {zeta=:.4f}, {wnd1=:.4f}, {wnd2=:.4f}')

overdamped(-31.67, -140.8)
overdamped(-2.121, -2.983)
```

gives

```
wn=66.7768, zeta=1.2914, wnd1=44.5865, wnd2=54.5650
wn=2.5153, zeta=1.0146, wnd1=0.0739, wnd2=0.4310
```

Nabil-Bishtawi, 2023-01-21:
I hope this will clarify the issue

[image]

ilayn, 2023-01-21:
@Nabil-Bishtawi Damping computation relies heavily on the classical first or second order system interpretation. When you have the system order larger than two, you are basically hoping to separate things into second and first order systems. You can still do this but you can't take the analogy too far. 

Especially if you have more than two real axis poles you don't know which one should be grouped with which pole to assume overdamped 2nd order system. For example, if you have poles (-1, -4, -4.4) how do you decide? 

That's why damp and other natural frequency computations (including matlab) is done per pole basis. They don't give too much information about the system anyways but it is typically used in the educational context. For this reason the algorithm for computing damping is the `-cos(angle of the pole)` (See bottom table https://www.mathworks.com/help/control/ref/lti.damp.html) and hence capped at 1 because if it is real then damping number is not meaningful.