python-control/python-control #848: create_statefbk_iosystem sets system name incorrectly

murrayrm, 2023-01-14:
The `create_statefbk_iosystem` sets the name of the closed loop system that it creates to one of the system inputs.  For example:
```
import numpy as np
import control as ct
sys = ct.ss([[0, 10], [-1, 0]], [[0], [1]], np.eye(2), 0, name='sys')
K, _, _ = ct.lqr(sys, np.eye(sys.nstates), np.eye(sys.ninputs))
ctrl, clsys = ct.create_statefbk_iosystem(sys, K)
print(clsys)
```
gives
```
<LinearICSystem>: u[0]
Inputs (3): ['xd[0]', 'xd[1]', 'ud[0]']
Outputs (3): ['y[0]', 'y[1]', 'u[0]']
States (2): ['sys_x[0]', 'sys_x[1]']
```
which has the name of the system as `u[0]` instead of something like `sys[3]`.

henklaak, 2023-02-01:
Problem is in the iosys.py 2871, where 'name' variable gets clobbered.
Can't be arsed to make a Pull Request, because Github forces me to jump too many hoops with tokens.

henklaak, 2023-02-02:
Close?

bnavigator, 2023-02-02:
Closed by #857

closed 2023-02-02 by bnavigator