python-control/python-control #856: import control segfaults on darwin-arm64

bdube-jpl, 2023-01-31:
installed via conda, with all the latest numpy, etc, explicitly darwin-arm64 versions.  Also happens on 3.10.8
```
$ python
Python 3.11.0 | packaged by conda-forge | (main, Jan 14 2023, 12:26:40) [Clang 14.0.6 ] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import control
[1]    3606 segmentation fault  python
```

Version is `control            conda-forge/noarch::control-0.9.3.post2-pyhd8ed1ab_0`

murrayrm, 2023-02-01:
Can you try `import slycot` and see if that is the source of the problem?

Also: what version of MacOS are you using?

bnavigator, 2023-02-01:
Possibly another case of https://github.com/python-control/Slycot/issues/186.

Do we have reports of functioning Mac on Arm installs? @murrayrm, @sawyerbfuller you have Macs right?

bdube-jpl, 2023-02-01:
This is on Ventura 13.1, 16 inch M1 Pro machine

import slycot does indeed also segfault; if this is covered under Slycot#186, feel free to close this issue

bnavigator, 2023-02-01:
Please feel free to report details in https://github.com/python-control/Slycot/issues/186. Particularly if you can post core-dumps, debugger backtraces and results for a self-compiled Slycot instead of the conda-forge package.

closed 2023-02-01 by bnavigator