python-control/python-control #866: Numpy 1.24.2 breaks  OS/BLAS test matrix

henklaak, 2023-02-18:
```
----------------------------- Captured stdout call -----------------------------
Message: Iteration limit reached
Platform: Linux-5.15.0-1033-azure-x86_64-with-glibc2.35
Python: 3.11.0
NumPy version: 1.24.2
blas_info:
    libraries = ['cblas', 'blas', 'cblas', 'blas']
    library_dirs = ['/usr/share/miniconda3/envs/test-env/lib']
    include_dirs = ['/usr/share/miniconda3/envs/test-env/include']
    language = c
    define_macros = [('HAVE_CBLAS', None)]
blas_opt_info:
    define_macros = [('NO_ATLAS_INFO', 1), ('HAVE_CBLAS', None)]
    libraries = ['cblas', 'blas', 'cblas', 'blas']
    library_dirs = ['/usr/share/miniconda3/envs/test-env/lib']
    include_dirs = ['/usr/share/miniconda3/envs/test-env/include']
    language = c
lapack_info:
    libraries = ['lapack', 'blas', 'lapack', 'blas']
    library_dirs = ['/usr/share/miniconda3/envs/test-env/lib']
    language = f77
lapack_opt_info:
    libraries = ['lapack', 'blas', 'lapack', 'blas', 'cblas', 'blas', 'cblas', 'blas']
    library_dirs = ['/usr/share/miniconda3/envs/test-env/lib']
    language = c
    define_macros = [('NO_ATLAS_INFO', 1), ('HAVE_CBLAS', None)]
    include_dirs = ['/usr/share/miniconda3/envs/test-env/include']
Supported SIMD extensions in this NumPy install:
    baseline = SSE,SSE2,SSE3
    found = SSSE3,SSE41,POPCNT,SSE42,AVX,F16C,FMA3,AVX2,AVX512F,AVX512CD,AVX512_SKX
    not found = AVX512_KNL,AVX512_KNM,AVX512_CLX,AVX512_CNL,AVX512_ICL
JOBNAME: conda ubuntu-3.11 Generic
```

henklaak, 2023-02-18:
In flatsys_tes, test_kinematic_car_ocp() an exception is made for specific numpy versions 1.24.0 and 1.24.1
Probably need to rethink that mechanism.

murrayrm, 2023-03-25:
The reason for making the check explicit to the version is so that we can track when there is an issue on a specific OS/BLAS/NumPy version, as is the case here.  I agree that it would be better to create a list of known exceptions associated with this workflow rather than having to update the individual unit tests.

PR #880 (2023-03-31): Small fixes for 0.9.4 release
This PR implements a number of small fixes for issues tagged for resolution in release 0.9.4:

* Updated installation documentation to recommend conda installation first, then pip (resolves #850)
* Added more documentation about Nyquist plot scaling (resolves #830)
* Deprecated `issys()` (resolves #876)
* Changed short package name for `control.optimal` from `opt` to `obc` consistently throughout documentation.
* Fixed sphinx processing of version to allow proper links to source code (resolves #837).
* Added numpy 1.24.2 to list of exceptions for flatness test on  [... cut]
closed 2023-03-31 by murrayrm