python-control/python-control #896: Can't connect an InterconnectedSystem to another system: "Couldn't parse input signal reference"

joaoantoniocardoso, 2023-05-25:
Let's say I have an InterconnectedSystem like so:
```Python
ab = ct.interconnect(
    [
        ct.rss(name='a', inputs=('u0'), outputs=('y0'), strictly_proper=True),
        ct.rss(name='b', inputs=('u0','u1'), outputs=('y0','y1'), strictly_proper=True),
    ],
    name='ab',
    connections=(
        ['b.u0', 'a.y0'],
    ),
    inputs=(
        'a.u0', 'b.u1',
    ),
    outputs=(
        'b.y0','b.y1',
    )
)
```
the output of its .__dict__ is:
```Python
{'name': 'ab',
 'ninputs': 2,
 'input_index': {'a.u0': 0, 'b.u1': 1},
 'noutputs': 2,
 'output_index': {'b.y0': 0, 'b.y1': 1},
 'nstates': 2,
 'state_index': {'a_x[0]': 0, 'b_x[0]': 1},
 'dt': 0,
 'params': {},
 'syslist': [<LinearIOSystem:a:['u0']->['y0']>,
  <LinearIOSystem:b:['u0', 'u1']->['y0', 'y1']>],
 'syslist_index': {'a': 0, 'b': 1},
 'state_offset': [0, 1],
 'input_offset': [0, 1],
 'output_offset': [0, 1],
 'connect_map': array([[0., 0., 0.],
        [1., 0., 0.],
        [0., 0., 0.]]),
 'input_map': array([[1., 0.],
        [0., 0.],
        [0., 1.]]),
 'output_map': array([[0., 1., 0., 0., 0., 0.],
        [0., 0., 1., 0., 0., 0.]]),
 'A': array([[ -0.51428322,   0.        ],
        [  0.        , -45.48019625]]),
 'B': array([[ 1.71550684,  0.        ],
        [ 0.        , -1.24010681]]),
 'C': array([[ 0.        ,  1.98391172],
        [ 0.        , -0.13830157]]),
 'D': array([[0., 0.],
        [0., 0.]])}
```

When I try to use this InterconnectedSystem as a subsystem of another InterconnectedSystem, I can't make the connections using the named syntax:
```Python
abc = ct.interconnect(
    [
        ab,
        ct.rss(name='c', inputs=('u0'), outputs=('y0'), strictly_proper=True),
    ],
    name='abc',
    connections=(
        ['ab.b.u1', 'c.y0'],  # <<<< ValueError: Couldn't parse input signal reference 'ab.b.u1'.
    ),
)
```

But if I left it unconnected, I can see the following warning, which clearly indicates that `ab.b.u1` should be a valid name, thus, it's probably a bug:

```Python
.env/lib/python3.10/site-packages/control/iosys.py:1503: UserWarning: Unused input(s) in InterconnectedSystem: (0, 1)=ab.b.u1; (1, 0)=c.u0; (0, 0)=ab.a.u0
  warn(msg)
.env/lib/python3.10/site-packages/control/iosys.py:1509: UserWarning: Unused output(s) in InterconnectedSystem: (0, 1) : ab.b.y1; (1, 0) : c.y0; (0, 0) : ab.b.y0
  warn(msg)
```

For more info, the output of `abc.__dict__` is:
```Python
{'name': 'abc',
 'ninputs': 0,
 'input_index': {},
 'noutputs': 0,
 'output_index': {},
 'nstates': 3,
 'state_index': {'ab_a_x[0]': 0, 'ab_b_x[0]': 1, 'c_x[0]': 2},
 'dt': 0,
 'params': {},
 'syslist': [<LinearICSystem:ab:['a.u0', 'b.u1']->['b.y0', 'b.y1']>,
  <LinearIOSystem:c:['u0']->['y0']>],
 'syslist_index': {'ab': 0, 'c': 1},
 'state_offset': [0, 2],
 'input_offset': [0, 2],
 'output_offset': [0, 2],
 'connect_map': array([[0., 0., 0.],
        [0., 0., 0.],
        [0., 0., 0.]]),
 'input_map': array([], shape=(3, 0), dtype=float64),
 'output_map': array([], shape=(0, 6), dtype=float64),
 'A': array([[ -0.51428322,   0.        ,   0.        ],
        [  0.        , -45.48019625,   0.        ],
        [  0.        ,   0.        ,  -1.59499572]]),
 'B': array([], shape=(3, 0), dtype=float64),
 'C': array([], shape=(0, 3), dtype=float64),
 'D': array([], shape=(0, 0), dtype=float64)}
```

Thanks

joaoantoniocardoso, 2023-05-25:
I'm 100000% open to workaround ideas

murrayrm, 2023-05-25:
Nested naming isn't supported yet and using a period-separated signal name shouldn't be allowed (that needs to be fixed).  The way to do what you are trying to do is to use `inplist` and `outlist` to identify the signals you want and then name them using `inputs` and `outputs`:
```
ab = ct.interconnect(
    [
        ct.rss(name='a', inputs=('u0'), outputs=('y0'), strictly_proper=True),
        ct.rss(name='b', inputs=('u0','u1'), outputs=('y0','y1'), strictly_proper=True),
    ],
    name='ab',
    connections=[
        ['b.u0', 'a.y0'],
    ],
    inplist=['a.u0', 'b.u1'],
    inputs=['a_u0', 'b_u1'],
    outlist=['b.y0', 'b.y1'],
    outputs=['b_y0', 'b_y1'],
)
```
then
```
abc = ct.interconnect(
    [
        ab,
        ct.rss(name='c', inputs=('u0'), outputs=('y0'), strictly_proper=True),
    ],
    name='abc',
    connections=[
        ['ab.b_u1', 'c.y0'],
    ],
)

closed 2023-06-04 by murrayrm