python-control/python-control #898: Implement *canon* function in MATLAB control toolbox

Jordan-Haidee, 2023-05-26:
In MATLAB, I can use *canon* func to convert common ss model to *Controllable Canonical Form* or *Observable Fanonical Form*，but in python-control repo it seems no similiar function since then, but I really need it.💖

sdahdah, 2023-05-26:
I think these are what you're looking for:

- https://python-control.readthedocs.io/en/0.9.3.post2/generated/control.canonical_form.html
- https://python-control.readthedocs.io/en/0.9.3.post2/generated/control.reachable_form.html
- https://python-control.readthedocs.io/en/0.9.3.post2/generated/control.observable_form.html#

Jordan-Haidee, 2023-05-26:
Thanks! It looks like I'm a dummy🤣.hahaha

 

  
https://python-control.readthedocs.io/en/0.9.3.post2/generated/control.canonical_form.html
 
https://python-control.readthedocs.io/en/0.9.3.post2/generated/control.reachable_form.html
 
https://python-control.readthedocs.io/en/0.9.3.post2/generated/control.observable_form.html#

murrayrm, 2023-05-26:
Reopening as an enhancement: at the least, the MATLAB compatibility library should have a `canon` function, if it doesn't already.