python-control/python-control #927: `TestFlatSys.test_kinematic_car_ocp[basis6-None-constraints6-None]` fails on 4-core VM

bmwiedemann, 2023-07-31:
While working on reproducible-builds for openSUSE, I noticed a semi-random failure in our `python-control` 0.9.4 package.

full build log: https://www.zq1.de/~bernhard/temp/python-control-build-log.txt

murrayrm, 2024-11-23:
Not sure if this is still an open issue.  Closing it out for now, but if you are still seeing this problem in v0.10.1 or above, please re-open.

closed 2024-11-23 by murrayrm

bmwiedemann, 2024-11-23:
There is now a different test failing in 0.10.1 (in 1 of 2 runs):
https://rb.zq1.de/other/python-control/build-log-0.10.1.txt
`FAILED control/tests/iosys_test.py::test_nonuniform_timepts[2-1-1] - Assertio...`

You might be able to reproduce it on Debian or openSUSE with
```bash
osc checkout openSUSE:Factory/python-control && cd $_
osc build --vm-type=kvm --noservice --clean -j4 standard
```