python-control/python-control #935: tf2ss() returns unstable system

pooyasf, 2023-10-17:
I have the following transfer function:
`_sys1.den=[[array([9.43513863e-11, 6.05312352e-08, 7.92752628e-07, 5.23764693e-06,
         1.82502556e-05, 1.24355899e-05, 8.68206174e-06, 2.73818482e-06,
         4.29133144e-07, 3.85554417e-08, 1.62631575e-09, 8.41098151e-12,
         9.85278302e-15, 4.07646645e-18, 5.55496497e-22, 3.06560494e-26,
         5.98908988e-31])]]`
'_sys1.num=[[array([9.94004350e-13, 2.67602795e-11, 2.31058712e-10, 1.15119493e-09,
         5.04635153e-09, 1.34066064e-08, 2.11938725e-08, 2.39940325e-08,
         2.05897777e-08, 1.17092854e-08, 4.71236875e-09, 1.19497537e-09,
         1.90815347e-10, 1.00655454e-11, 1.47388887e-13, 8.40314881e-16,
         1.67195685e-18])]]'

asking for poles:

`_sys1.poles()=array([-6.28318525e+02+0.00000000e+00j, -6.28318531e+00+0.00000000e+00j,
       -3.14159265e+00+5.44139809e+00j, -3.14159265e+00-5.44139809e+00j,
       -1.19380521e-01+5.84842723e-01j, -1.19380521e-01-5.84842723e-01j,
       -1.88495559e-01+0.00000000e+00j, -6.63225116e-02+9.92624461e-02j,
       -6.63225116e-02-9.92624461e-02j, -1.00530965e-01+0.00000000e+00j,
       -4.32233973e-03+0.00000000e+00j, -6.28318476e-04+0.00000000e+00j,
       -6.28318592e-04+0.00000000e+00j, -6.29820167e-05+2.60763321e-07j,
       -6.29820167e-05-2.60763321e-07j, -6.25315230e-05+0.00000000e+00j])`

All negative. System is stable. But then when converting _sys1 to state space using tf2ss() returns a system which have positive poles (unstable). I am using Slycot.

I also repeated the same conversion but using scipy and it gives me a stable state space as a result.

python-control version: '0.9.3.post2'

murrayrm, 2023-10-19:
This looks like a bug in SLICOT, since `tf2ss` (which eventually calls `statesp._convert_to_statespace`) just calls the SLICOT `td04ad` routine.  Using `scipy.signal.tf2ss` seems to give the correct answer.

Needs a closer took to make sure there is not a python-control bug someplace.  If not, we should transfer to the slycot issues list.

murrayrm, 2023-10-22:
I've created a slycot issue ([#222](https://github.com/python-control/Slycot/issues/222)) reporting this issue.  PR coming that includes a unit test for this case and also adds a `method` keyword to `tf2ss`, plus some documentation pointing out the new functionality and the slycot bug.

murrayrm, 2024-11-23:
Closing this out since this is no longer a python-control issue.

closed 2024-11-23 by murrayrm