python-control/python-control #944: control.bode(Hz=True) the conversion is wrong.

fermvikt, 2023-12-05:
Hi,

Version: control 0.9.4
\control\freqplot.py

Line 320:

                if Hz:
                    # omega_plot = omega_sys / (2. * math.pi)
                    omega_plot = omega_sys * (180 / (math.pi))

Line: 386

                    if Hz:
                        # Wcg, Wcp = Wcg/(2*math.pi), Wcp/(2*math.pi)
                        Wcg, Wcp = Wcg * (180 / (math.pi)), Wcp * (180 / (math.pi))

Best regards,
Vikt

bnavigator, 2023-12-05:
The frequency is `f = 2*pi*omega` with `f` in `Hz` and `omega` in `rad/s` . These are not angles in degrees.

Side note: we work on code in the `main` branch, not in a released version.

closed 2023-12-05 by bnavigator

fermvikt, 2023-12-05:
Sorry,
Just for clarification.

                    Wcg, Wcp = Wcg/(2*math.pi), Wcp/(2*math.pi)

If this line is supposed to convert Wcg, Wcp from rad/s to Hz, I would do it by multiplying by (180/pi) and not by dividing by 2pi. 
Am I missing something here?

Best regards,
Viktor Ferm

fermvikt, 2023-12-05:
Okay, thank you.