python-control/python-control #949: Different time response using TF or IO system representation

josujugo, 2023-12-23:
Hi all:

I have made a simple test  using the "input/output systems" functions for converting a TF into a IO system and I have obtained a different time response using the original TF and the IO system for a sinusoidal input signal. The code is this one:

import control as ct
import matplotlib.pyplot as plt
import numpy as np
w0=1000
K=0.1
Q0=1000

s=ct.tf('s')
cavity = K*w0**2/(s**2+w0/Q0*s+w0**2)

t=np.r_[0:10:0.0001]
y=ct.forced_response(cavity,t,np.sin(w0*t))    
plt.plot(y.time,y.y[0,:])
io_cavity=ct.tf2io(cavity)
t, y = ct.input_output_response(io_cavity, t, np.sin(w0*t))
plt.plot(t,y)

Maybe, I am interpreting something wrongly.

murrayrm, 2023-12-24:
The difference in the outputs is happening because `forced_response` and `input_output_response` use different methods to carry out the simulation:

* `forced_response` is intended for use with linear systems and does an exact calculation of the exponential response of the (time-discretized) input.  (If you call `forced_response` with a nonlinear I/O system, it will call `input_output_response`.)
* `input_output_response` works on linear or nonlinear systems and calls the SciPy `solve_ivp` function.

The reason you are getting different responses is because of the default settings for `solve_ivp`, which uses the `RK45` integrator.  Since you are simulating a high-Q system right at the resonant frequency, the default integrator is not very accurate.  If you re-run using `LSODA` as the integrator, it looks to me like things line up:

    t, y = ct.input_output_response(io_cavity, t, np.sin(w0 * t), solve_ivp_method='LSODA')

murrayrm, 2023-12-24:
The following call to `input_output_response` gives results that are very close:
```
t, y_io = ct.input_output_response(
     io_cavity, t, np.sin(w0*t), solve_ivp_kwargs={'rtol': 1e-5})
```
(the default `rol` for `RK45` is 1e-3).

closed 2023-12-24 by murrayrm

josujugo, 2023-12-24:
Thank you very  much @murrayrm. I was sure that I was missing something. 

Happy new year!