python-control/python-control #950: gain offset using control.matlab.c2d using method='matched' ?

JonHowMIT, 2023-12-24:
Compared results of 
sysc = ct.TransferFunction([1, z], [1, p])
sysd = ct.matlab.c2d(sysc, Ts, method='bilinear')
sysd1 = ct.matlab.c2d(sysc, Ts, method='matched')

yields a sysd1 with a gain that is about 10x that of sysd - is that expected?

sawyerbfuller, 2023-12-25:
Hi John great to hear from you and happy holidays. Glad to see you are using or at the very least evaluating the library. 

The matched c2d is code I have never used before and it doesn’t look like we have a unit test for it, so there is a decent chance it is just a bug. Thanks for the report! 

One q - when you refer to gain, do you mean the DC gain of the two systems?

murrayrm, 2023-12-26:
I did a quick check and I agree there is something wrong in the matched transformation:
```
import control as ct
sysc = ct.tf([1, 2], [1, 3], name='sysc')
sysd_bilinear = ct.sample_system(sysc, 0.1, method='bilinear', name='sysd_bilinear')
sysd_matched = ct.sample_system(sysc, 0.1, method='matched', name='sysd_matched')
ct.bode([sysd_bilinear, sysd_matched, sysc])
```

This produces the following plot:
[image]

A couple of things to fix:
* Zero frequency gain for the 'matched' transformation is incorrect.
* The name of the system is lost when doing the 'matched' transformation.
* The order of the system names is incorrect in the figure title

murrayrm, 2023-12-26:
Fixed in #951.  New plot:

[image]

closed 2023-12-26 by murrayrm