python-control/python-control #954: TypeError: No support for MIMO without slycot

hariharan382, 2023-12-29:
raise TypeError("No support for MIMO without slycot")
TypeError: No support for MIMO without slycot

I am trying following example ,

sys_mimo = ct.tf2ss(
    [[[1], [0.1]], [[0.2], [1]]],
    [[[1, 0.6, 1], [1, 1, 1]], [[1, 0.4, 1], [1, 2, 1]]], name="sys_mimo")
response = ct.step_response(sys)
response.plot()

environment :
  conda env
* slycot                    0.5.4       
* control                   0.9.4

I have already installed the slycot , as I could see it "conda list" command, but I have no way the script is complaining no slycot module

murrayrm, 2023-12-30:
A couple of things to try:
* Using python, ipython, Jupyter or something equivalent, make sure that `import slycot` runs and that `slycot.__version__` is 0.5.4
* If that works, run `import control as ct` and check that `ct.slycot_check()` returns `True`.

hariharan382, 2024-01-03:
I had to try mamba , tried different version of python in conda , completely helpless

murrayrm, 2024-01-03:
What were the results of the two tests I suggested above?

hariharan382, 2024-01-05:
yeah ,slycot was not installing , So I had to switch to mamba to solve this issue,

I used following environment setup it is working 

`mamba create -n <new-environment> pytest numpy scipy control matplotlib ipykernel slycot python=3.12`

closed 2024-01-05 by hariharan382