python-control/python-control #968: `phase_plot` ODE function signature is not compatible with `solve_ivp` and `StateSpace.dynamics`

FeldrinH, 2024-02-09:
There seems to be some inconsistency with the ordering of parameters for ODE functions between `phase_plot` and `StateSpace.dynamics`. `phase_plot` expects a function with signature f(x, t), whereas `StateSpace.dynamics` and other functions that are meant to be used with `scipy.integrate.solve_ivp` provide a function with the signature f(t, x) (note that the order of arguments is different).

It would be nice if the order of arguments was consistent, so you could pass functions compatible with `scipy.integrate.solve_ivp` directly to `phase_plot`.

sawyerbfuller, 2024-03-05:
Yes I agree, `phase_plot` has the wrong parameter order, leftover from the previous convention. The docstring should also be updated to reference scipy's new integration scheme as well in `scipy.integrate.solve_ivp`. 

Please submit a [pull request](https://github.com/python-control/python-control/wiki/How-to-contribute-with-a-pull-request)!

FeldrinH, 2024-03-05:
@murrayrm commented in https://github.com/python-control/python-control/issues/970#issuecomment-1951375460 that he is rewriting parts of `phase_plot`, so it might not be worthwhile for an external contributor to do a pull request right now.

closed 2024-03-31 by murrayrm