python-control/python-control #973: `control.mixsyn` hangs on M1

NaoPross, 2024-02-29:
## Description

I have the following code in a file `bug.py`
```python3
import control as ct

G = ct.tf([1, 1], [1, 0.5, 4])
Wp = ct.tf([2, -2.2, 1], [3, 0.2, 0.01])

k, cl, p, info = ct.mixsyn(G, Wp)
```
and when I run it it seems to hang forever. Compare this to the MATLAB equivalent
```matlab
G = tf([1, 1], [1, 0.5, 4])
Wp = tf([2, -2.2, 1], [3, 0.2, 0.01])

[K,CL,GAM,info] = mixsyn(G, Wp)
```
which runs fine.

<details>
<summary>Expand for MATLAB output</summary>

```
G =
 
       s + 1
  ---------------
  s^2 + 0.5 s + 4
 
Continuous-time transfer function.

Wp =
 
   2 s^2 - 2.2 s + 1
  --------------------
  3 s^2 + 0.2 s + 0.01
 
Continuous-time transfer function.

K =
 
  A = 
               x1          x2          x3          x4
   x1    -0.06667    -0.05333           0           0
   x2      0.0625   2.233e-31  -5.716e-32  -2.849e-32
   x3       205.1        1574      -398.1      -197.9
   x4  -2.098e-15  -1.612e-14           2   2.056e-15
 
  B = 
               u1
   x1           2
   x2  -9.533e-34
   x3  -8.344e-17
   x4   6.879e-17
 
  C = 
           x1      x2      x3      x4
   y1   205.1    1574  -397.6  -195.9
 
  D = 
       u1
   y1   0
 
Continuous-time state-space model.

CL =
 
  A = 
               x1          x2          x3          x4          x5
   x1    -0.06667    -0.05333          -2          -1           0
   x2      0.0625           0           0           0           0
   x3           0           0        -0.5          -2       205.1
   x4           0           0           2           0           0
   x5           0           0          -2          -1    -0.06667
   x6           0           0   9.533e-34   4.767e-34      0.0625
   x7           0           0   8.344e-17   4.172e-17       205.1
   x8           0           0  -6.879e-17  -3.439e-17  -2.098e-15
 
               x6          x7          x8
   x1           0           0           0
   x2           0           0           0
   x3        1574      -397.6      -195.9
   x4           0           0           0
   x5    -0.05333           0           0
   x6   2.233e-31  -5.716e-32  -2.849e-32
   x7        1574      -398.1      -197.9
   x8  -1.612e-14           2   2.056e-15
 
  B = 
               u1
   x1           2
   x2           0
   x3           0
   x4           0
   x5           2
   x6  -9.533e-34
   x7  -8.344e-17
   x8   6.879e-17
 
  C = 
            x1       x2       x3       x4       x5       x6       x7
   y1  -0.3889    2.649  -0.6667  -0.3333        0        0        0
 
            x8
   y1        0
 
  D = 
           u1
   y1  0.6667
 
Input groups:       
    Name    Channels
     U1        1    
                    
Output groups:      
    Name    Channels
     Y1        1    
                    
Continuous-time state-space model.

GAM =

    0.6686

info = 

  hinfINFO with properties:

    gamma: 0.6733
        X: [4×4 double]
        Y: [4×4 double]
       Ku: [205.0868 1.5738e+03 -397.5912 -195.9132]
       Kw: [30.4970 234.2721 -58.9574 -29.3820]
       Lx: [4×1 double]
       Lu: 0
     Preg: [3×2 ss]
       AS: [2×2 ss]
```
</details>

## Hardware & Versions

- Apple MacBook M1 Pro with macOS 13.5 (22G74)
- Python 3.11.7, and `control.__version__` is `0.9.3.post2` and `slycot.__version__` is `0.5.4`
- MATLAB R2022b Update 5, 9.13.0.2193358 (maci64) pretty sure it runs on Rosetta

## Trace

I tried to run the code with
```
$ python3 -m trace --trace bug.py
```
and the code hangs (forever?) at
```
...
synthesis.py(1306):         LW7 = m2*np2 + np2*np2 + m2*m2 + max(nd1*nd1 + max(2*nd1, (nd1 + nd2)*np2),\
synthesis.py(1309):         ldwork = LW1 + max(1, LW2, LW3, LW4, LW5 + max(LW6,LW7))
synthesis.py(1310):     out = _wrapper.sb10ad(job,n,m,np,ncon,nmeas,gamma,A,B,C,D,gtol,actol,liwork,ldwork)
```
I'll try to run this again later on an x86 machine, but I suspect this problem is only on M1.

NaoPross, 2024-02-29:
Related:  #152 #367

Ok, it seems this is a known problem of `sb10ad`. In the past I've had convergence issues with other algorithms because of the new arm architecture, so I was suspecting this is an M1 problem, but it seems like it's unrelated.

murrayrm, 2024-03-02:
Tagging this as a duplicate of #152.

@NaoPross You might consider posting this issue to the developers of SLICOT (which is what slycot is based on): https://www.slicot.org

closed 2024-03-02 by murrayrm