python-control/python-control #991: Parallel system interconnection using `control.interconnect()`

sdahdah, 2024-04-10:
I'm trying to use `control.interconnect()` to connect one system input to two subsystem inputs, but I can't figure out a clean way to do it. The best way I've found so far is to create a dummy summing junction for this purpose, as shown below:

```python
import control

# Systems to connect in parallel
a = control.TransferFunction([1], [1], name='a')
b = control.TransferFunction([1], [2], name='b')
sum = control.summing_junction(2, name='sum')
split = control.summing_junction(1, name='split')

# Interconnection using ``control.interconnect``
P_interconnect = control.interconnect(
    syslist=[a, b, sum, split],
    connections=[
        ['sum.u[0]', 'a'],
        ['sum.u[1]', 'b'],
        ['a', 'split'],
        ['b', 'split'],
    ],
    inplist=['split'],
    outlist=['sum'],
)
print(P_interconnect)

# Interconnection using ``control.parallel``
P_parallel = control.parallel(a, b)
print(control.tf2ss(P_parallel))
```

Am I missing something, or is this the recommended way to feed two subsystem blocks the same system-level input?

slivingston, 2024-04-12:
`connections` specifies internal connections (i.e., between subsystems), whereas auto-splitting behavior only affects how system inputs map to subsystem inputs. Therefore, `connections` can be given when doing this.

Note that automatically (also known as "implicitly") making internal connections will not work with an explicit `connections` argument. For reference, the relevant part of the code is https://github.com/python-control/python-control/blob/1bb0a46d2475e1b185a4cb6f8c45f64aa0248526/control/nlsys.py#L2260-L2273

sdahdah, 2024-04-12:
Got it. Thanks for the help!

closed 2024-04-12 by sdahdah