python-control/python-control #992: Plotting nyquist plot from FrequencyResponseData fails

wueestry, 2024-04-16:
When trying to create the nyquist plot from an FRD system
```python
freq_response = ct.FRD(response, frequencies)
ct.nyquist_plot(freq_response)
```
the plotting fails with the following error
`AttributeError: 'FrequencyResponseData' object has no attribute 'poles'` during the call of `_default_frequency_range`

version: 0.10.0

slivingston, 2024-04-16:
Is it possible to make a Nyquist plot given only frequency response data?

murrayrm, 2024-04-17:
You can get a valid Nyquist plot from the frequency response in some instances and with some assumptions.  Basically, if the open loop dynamics are assumed stable, the zero frequency gain is finite, and the high frequency gain is zero, it *should* work.

The error above comes when it tries to compute the default frequency range for the plot, but in the case of frequency response data you don't actually need to do that.

A quick fix is to change the call to be `ct.nyquist_plot(freq_response, freq_response.omega)`, which bypasses the computation of the default frequency range.

murrayrm, 2024-06-13:
Fixed in #996.

closed 2024-06-13 by murrayrm