python-control/python-control #998: inconsistent kwarg klist/kvect/gains in rlocus doc

gdmcbain, 2024-05-09:
The first line of the docstring for `rlocus` suggests that there's a kwarg `klist`,

    """rlocus(sys[, klist, xlim, ylim, ...])
https://github.com/python-control/python-control/blob/e4a03e8419231b040d846ef089babd029e18168c/control/matlab/wrappers.py#L200

 but in the list of parameters it's `kvect`.  

https://github.com/python-control/python-control/blob/e4a03e8419231b040d846ef089babd029e18168c/control/matlab/wrappers.py#L212

They're all just passed on to `root_locus_plot` anyway 

https://github.com/python-control/python-control/blob/e4a03e8419231b040d846ef089babd029e18168c/control/matlab/wrappers.py#L247

and it's expecting `kvect`, or, really, `gains`

https://github.com/python-control/python-control/blob/e4a03e8419231b040d846ef089babd029e18168c/control/rlocus.py#L163

gdmcbain, 2024-05-09:
This is miscalled in pymoca/pymoca [test/notebooks/Spring,ipynb](https://github.com/pymoca/pymoca/blob/9d597d76aa2b82bd3d563d4938a098af8baafd49/test/notebooks/Spring.ipynb#L1265):

```python
    "control.rlocus(ss[0,0], klist=pl.logspace(-2,1,1000));\n",
```

bnavigator, 2024-05-09:
> The first line of the docstring for rlocus suggests that there's a kwarg klist,

Only for the MATLAB wrapper though. The MATLAB docs call the parameter `k` by the way.
Using a MATLAB wrapper with a python kwarg is somehow unusual.

PR welcome.

> This is miscalled in pymoca/pymoca [test/notebooks/Spring,ipynb](https://github.com/pymoca/pymoca/blob/9d597d76aa2b82bd3d563d4938a098af8baafd49/test/notebooks/Spring.ipynb#L1265):

The MATLAB wrapper is not part of that call. The kwarg for control.rlocus has been `kvect` for 13 years: 8ebb73b76b856b257a51c15b3c6eda2eac6e6c68.

gdmcbain, 2024-05-10:
Ah, yes, sorry I see now that [`control.rlocus`](https://github.com/python-control/python-control/blob/e4a03e8419231b040d846ef089babd029e18168c/control/rlocus.py#L452) doesn't call [`control.matlab.wrappers.rlocus`](https://github.com/python-control/python-control/blob/e4a03e8419231b040d846ef089babd029e18168c/control/matlab/wrappers.py#L200) at all but rather `control.root_locus_plot`; I had overlooked this:

https://github.com/python-control/python-control/blob/e4a03e8419231b040d846ef089babd029e18168c/control/rlocus.py#L452

My real problem was pymoca/pymoca#324 so I'll look over there at fixing that.  I'm not presently concerned with `control.matlab`, but I'll turn in the PR as requested, while I'm at it.

PR #999 (2024-05-11): klist->gains kwarg for matlab.wrappers.rlocus
This fixes #998.

It wasn't the actual problem there, with `control.matlab.wrappers.rlocus` not being called (rather `control.rlocus.rlocus = control.rlocus.root_locus_plot`), but still this does forward kwargs to that function which is expecting `gains` rather than `klist` or `kvect` (though it does have a `control._process_legacy_keyword` in place to catch the latter).

Feel free to ignore or use this.
closed 2024-05-11 by murrayrm