python-control/python-control #1002: How to convert finite step response model to state space representation?

csianglim, 2024-05-11:
I have some simulation data for a discrete LTI system which are provided as unit step response models between the 5 inputs (rows) and 12 outputs (columns).

I'd like to convert the model to a state space representation in `python-control` but I'm not sure how to do that.

Based on what I've read so far, it seems like the best way is to convert the step response models to impulse response first, then use `control.era` to get a state space representation. Am I on the right track?

[image]

Thanks,
Siang

murrayrm, 2024-05-12:
This functionality does not yet exist in the python-control toolbox (the `era` function is raises an "Not Implemented" exception).  It would be great if you wanted to contribute this functionality via a pull request!

csianglim, 2024-05-13:
Thanks, I'll give it a try.