python-control/python-control #1010: equations not showing up correctly in online docs

sawyerbfuller, 2024-06-11:
Looks like none of the equations are being rendered correctly at readthedocs.io. Example screenshot from `ss` at https://python-control.readthedocs.io/en/0.10.0/generated/control.ss.html#control.ss:

[image]

bnavigator, 2024-06-11:
Strange.

The images of version 0.10.0 are HTTP 404 Not found:
https://python-control.readthedocs.io/en/0.10.0/generated/_images/math/395f23ad1c23659a504fc6c29d7cc2b35aac2e3a.png

The rendering is fine for "latest" and "0.9.4"

https://python-control.readthedocs.io/en/latest/generated/control.ss.html
https://python-control.readthedocs.io/en/0.9.4/generated/control.ss.html

murrayrm, 2024-06-13:
There's some sort of a strange bug on readthedocs that requires redirecting image pages.  The fix is documented in the [release instructions](https://github.com/python-control/python-control/wiki/Release-instructions#deploy-release).

The redirect is in place for all other versions of python-control, but when I release 0.10.0 it was working without this, so I didn't add the redirect.  I guess something changed and it stopped working.

I posted an issue to the readthedocs github site ([here](https://github.com/readthedocs/readthedocs.org/issues/9852)), but it never really got resolved.

Anyway, I put the redirect in for 0.10.0 and things are working again.

closed 2024-06-13 by murrayrm