python-control/python-control #1014: IndexError: index -1 is out of bounds for axis 0 with size 0

NikolaiVChr, 2024-06-27:
I get this:
```
Traceback (most recent call last):
  File "C:\Users\me\PycharmProjects\f\control1.py", line 17, in <module>
    S = ct.step_info(sysdata=y_out2, T=T2)
  File "C:\Users\me\PycharmProjects\f\venv\lib\site-packages\control\timeresp.py", line 1614, in step_info
    settled = np.where(
IndexError: index -1 is out of bounds for axis 0 with size 0
```
From this: (just a simple integrator TF)
```python
import control as ct
import numpy as np

num = [0,5]
den = [1,0]

H = ct.TransferFunction(num, den)

t = np.arange(start=0, stop=2, step=1/120)

T2, y_out2 = ct.timeresp.impulse_response(sys=H, T=t)
pyplot.figure(1)
pyplot.plot(T2, y_out2)
pyplot.grid(True)
pyplot.show(block=False)
S = ct.step_info(sysdata=y_out2, T=T2)

pyplot.show()
```

Control v0.10.0
Numpy 1.26.4
Scipy 1.14.0

NikolaiVChr, 2024-06-27:
I get same error with:

num = [1,5]
den = [1,0]

NikolaiVChr, 2024-06-27:
Oh, and thank you for great library. Its awesome.

murrayrm, 2024-06-27:
The bug is in the `step_info` function.  The impulse response for a pure integrator is a constant output and `step_info` gets confused when it is passed something that doesn't look like a step response.

Fixing the bug should be straightforward: we should probably issue a warning when we can't find the features we are looking for in a typical step response.

slivingston, 2024-07-13:
Fixed in #1029

closed 2024-07-13 by slivingston