python-control/python-control #1015: "Interconnect" input specification.

bonidydy, 2024-06-29:
Hi,

In this code I want "interconnect" to build a system with just one input "sys.u1". It tries to create two inputs and as a result catches an exception. Am I getting it wrong?

```
import numpy as np
import control as ct

f = np.array([[-35, 1], [-225, -5]])
j = np.array([[0], [5]])
g = np.array([[-965], [-8775]])
cobs = np.eye(2)
dobs = np.zeros((2, 2))

a = np.array([[0, 1, 0], [0, -1, 1], [0, 0, -5]])
b = np.array([[0], [0], [5]])
c = np.array([[1, 0, 0]])
d = 0

plant = ct.ss(a, b, c, d,
              inputs=["u1"], outputs=["y1"], name="plant")
observer = ct.ss(f, np.hstack((g, j)), cobs, dobs,
                 inputs=["u1", "u2"], outputs=["y1", "y2"], name="observer")

sys = ct.interconnect([plant, observer],
                      connections=[["observer.u1", "plant.y1"]],
                      inplist=[["plant.u1", "observer.u2"]],
                      outlist=[["plant.y1"], ["observer.y1"], ["observer.y2"]],
                      inputs=["u1"],
                      outputs=["y1", "y2", "y3"],
                      debug=True)
```

murrayrm, 2024-06-29:
Looks like a bug in the way input (and output) lists are handled.  The code is not catching the fact that a single input is going to multiple subsystems.

Fix coming in a bit.

bonidydy, 2024-06-30:
It seems there is also a mistake in the documentation of "interconnect", "inplist" argument:

inplist (list of input connections, optional) –
List of connections for how the inputs for the overall system are mapped
to the subsystem inputs. The input specification is similar to the form
defined in the connection specification, except that connections do not
specify an **input-spec**, since these are the system inputs. The entries for a
connection are thus of the form:
[input-spec1, input-spec2, …]

I believe the bold text should be **output-spec**.

closed 2024-07-07 by murrayrm