python-control/python-control #1016: UnboundLocalError: local variable 's' referenced before assignment

NikolaiVChr, 2024-06-29:
When mouse clicking in root locus plot, I sometimes get this:
```
Traceback (most recent call last):
  File "C:\Users\me\PycharmProjects\f\venv\lib\site-packages\matplotlib\cbook.py", line 298, in process
    func(*args, **kwargs)
  File "C:\Users\me\PycharmProjects\f\venv\lib\site-packages\control\pzmap.py", line 470, in _click_dispatcher
    K, s = _find_root_locus_gain(event, sys, ax)
  File "C:\Users\me\PycharmProjects\f\venv\lib\site-packages\control\pzmap.py", line 531, in _find_root_locus_gain
    return None, s
UnboundLocalError: local variable 's' referenced before assignment
```

murrayrm, 2024-06-29:
This worked OK for me on the main branch with Python 3.12 and matplotlib 3.8.4 using iPython.

Can you post the versions of Python, control and matplotlib that you are using, along with the execution environment (python, iPython, Jupyter) that you are using, so we can narrow things down?

NikolaiVChr, 2024-06-30:
Control v0.10.0
Numpy 1.26.4
Scipy 1.14.0
matplotlib 2.9.0
PyCharm 2023.2.3

Python env

closed 2024-07-07 by murrayrm