python-control/python-control #1043: interconnect of named signals yields incorrect transfer function

allensnyder8, 2024-08-21:
I've found a strange situation where I do not get the correct transfer function when using named signals and interconnect. 

Using named signals I can create the correct transfer function given the pet problem of a second order low pass filter, but I could not yield the correct transfer function in a slightly more complex problem.  I was able to form the correct transfer function of the same system using old fashioned block diagram reduction.

It is possible I am not using the named signals approach correctly, but I cannot find an error in my code.

control version: 0.10.1
python version: 3.10.12 (main, Jul 29 2024, 16:56:48) [GCC 11.4.0]

[ProblemCode.txt](https://github.com/user-attachments/files/16685767/ExampleCode.txt)
[image]

murrayrm, 2024-08-22:
Here's the code (easier to see inline):
```
import control as ct
import numpy as np
import matplotlib.pyplot as plt

# Example second order transfer function
wn = 1
zeta = 0.707

b1 = ct.summing_junction(       inputs=['u','-y5'],     output='y1')
b2 = ct.tf(wn**2,1,             input = 'y1',           output='y2')
b3 = ct.summing_junction(       inputs=['y2','-y6'],    output='y3')
b4 = ct.tf(1,[1,0],             input='y3',             output='y4')
b5 = ct.tf(2*zeta*wn,1,         input='y4',             output='y6')
b6 = ct.tf(1,[1,0],             input='y4',             output='y5')

sys = ct.interconnect([b1,b2,b3,b4,b5,b6],inputs='u',outputs='y5')

ct.connection_table(sys)
#ct.frequency_response(sys).plot()

#### ^^ Returns the correct answer

### More complicated code

# Parameters
Jm = 3.2e-5 
Bm = 0.05466 / 160 
Kt = 0.411
gr = 1/160
wn = 50*2*np.pi
zeta = 0.06
Jl = 0.016
Kl = 1900
Bl =  2*zeta*np.sqrt(Kl*Jl)
s = ct.tf([1,0],[0,1])

##### Copy block diagram into code
#
# Surface Dynamics Block Diagram
b0 = ct.tf(1,[Jm,Bm],           input='Tn',             output='mv')
b1 = ct.tf(1,[1,0],             input='mv',             output='mp')
b2 = ct.tf(Kl,1,                input='dp',             output='Ti')
b3 = ct.tf(1,[Jl,Bl],           input='Ti',             output='lv')
b4 = ct.tf(1,[1,0],             input='lv',             output='lp')
b5 = ct.tf(Kt,1,                input='iq',             output='Tm')
b6 = ct.tf(gr,1,                input='Ti',             output='Tl')
b7 = ct.summing_junction(       inputs=['ap','-lp'],    output='dp')
b8 = ct.summing_junction(       inputs=['Tm','-Tl'],    output='Tn')

# Plot Motor Current to Motor Velocity
sys = ct.interconnect([b0,b1,b2,b3,b4,b5,b6,b7,b8],inputs='iq',outputs='mv')
ct.connection_table(sys)
ct.frequency_response(sys).plot()

print(' ')

# Plot Motor Curent to output velocity
sys = ct.interconnect([b0,b1,b2,b3,b4,b5,b6,b7,b8],inputs='iq',outputs='lv')
ct.connection_table(sys)
ct.frequency_response(sys).plot()

#### ^^^ None of the above transfer functions are correct

#########
# Do it the old fashioned way. 
#   Returns the correct answer
gm = ct.tf(1,[Jm,Bm])
gl = ct.tf(1,[Jl,Bl,0])
glf = ct.feedback(Kl,gl)
sys = Kt * ct.feedback(gm,gr/s*glf*gr)
IqToMotorVelocity = ct.series(sys,1,name='IqToMotorVelocity')
ct.frequency_response(IqToMotorVelocity).plot()
ct.damp(IqToMotorVelocity,doprint=True)
```

murrayrm, 2024-08-22:
When I run the code above, I get the following error:
```
/Users/murray/Library/CloudStorage/Dropbox/macosx/src/python-control/murrayrm/control/nlsys.py:1192: UserWarning: Unused input(s) in InterconnectedSystem: (7, 0)=sys[47].ap
  warn(msg)
/Users/murray/Library/CloudStorage/Dropbox/macosx/src/python-control/murrayrm/control/nlsys.py:1198: UserWarning: Unused output(s) in InterconnectedSystem: (1, 0) : sys[41].mp
  warn(msg)
```
It looks like you have the output from `b1` as 'mp' when you actually mean 'ap'.

On its own that did not make the frequency responses match, but I didn't trace down whether the block diagram algebra at the bottom is correct.

allensnyder8, 2024-08-22:
Ah.  You are correct it was user error, I also had an incorrect gain in b1.  The transfer functions match with the changes.  IT was all user error.

Thank you @murrayrm !

Fixed code:
```
import control as ct
import numpy as np
import matplotlib.pyplot as plt

# Parameters
Jm = 2.35e-5 
Bm = 0.05466 / 160 
Kt = 0.411
gr = 1/160
wn = 50*2*np.pi
zeta = 0.06
Jl = 0.016
Kl = 1500
#Kl = np.sqrt(wn/Jl)
Bl =  2*zeta*np.sqrt(Kl*Jl)
s = ct.tf([1,0],[0,1])

##### Copy block diagram into code
#
# Surface Dynamics Block Diagram
b0 = ct.tf(1,[Jm,Bm],           input='Tn',             output='mv')
b1 = ct.tf(gr,[1,0],            input='mv',             output='ap')
b2 = ct.tf(Kl,1,                input='dp',             output='Ti')
b3 = ct.tf(1,[Jl,Bl],           input='Ti',             output='lv')
b4 = ct.tf(1,[1,0],             input='lv',             output='lp')
b5 = ct.tf(Kt,1,                input='iq',             output='Tm')
b6 = ct.tf(gr,1,                input='Ti',             output='Tl')
b7 = ct.summing_junction(       inputs=['ap','-lp'],    output='dp')
b8 = ct.summing_junction(       inputs=['Tm','-Tl'],    output='Tn')

# Plot Motor Current to Motor Velocity
sys = ct.interconnect([b0,b1,b2,b3,b4,b5,b6,b7,b8],inputs='iq',outputs='mv',name='IqToMotorVelocity-interconnect')
ct.connection_table(sys)
ct.frequency_response(sys).plot(Hz=True)

print(' ')

# Plot Motor Curent to output velocity
#sys = ct.interconnect([b0,b1,b2,b3,b4,b5,b6,b7,b8],inputs='iq',outputs='lv',name='IqToLoadPosition')
#ct.connection_table(sys)
#ct.frequency_response(sys).plot(Hz=True)

#### ^^^ None of the above transfer functions are correct

#########
# Do it the old fashioned way. 
#   Returns the correct answer
gm = ct.tf(1,[Jm,Bm])
gl = ct.tf(1,[Jl,Bl,0])
glf = ct.feedback(Kl,gl)
sys = Kt * ct.feedback(gm,gr/s*glf*gr)
IqToMotorVelocity = ct.series(sys,1,name='IqToMotorVelocity-BlockDiagramReduction')
ct.frequency_response(IqToMotorVelocity).plot(Hz=True,linestyle=':')
ct.damp(IqToMotorVelocity,doprint=True)
```

[image]

closed 2024-08-22 by allensnyder8