python-control/python-control #1055: Installtion for arch linux fails (unable to find setup.py)

vibhoregit, 2024-11-11:
I am trying to install the control module on my PC running Manjaro Linux. I tried installing the control package using yay like so

`yay -S python-control-git`

Ended up with the following error:

==> Starting build()...
python: can't open file '/home/vibe/.cache/yay/python-control-git/src/python-control/setup.py': [Errno 2] No such file or directory
==> ERROR: A failure occurred in build().
Aborting...
-> error making: python-control-git-exit status 4
-> Failed to install the following packages. Manual intervention is required:
python-control-git - exit status 4

I understand that setup tool based installation is deprecated but how is yay supposed to go about installation or in general Arch linux installation proceed given that pacman does not recognize control module as official repo.

murrayrm, 2024-11-11:
I'm not familiar with Arch linux.  Is pip supported?  If so, that will be the easiest way to install the vanilla control package.

What will likely be harder is slycot, since this requires a FORTRAN compiler and (usually) lots of work getting the right libraries in place.  But you can do most SISO operations without slycot...

vibhoregit, 2024-11-11:
pip is not recommended for system wide installations. It may lead to broken dependencies.

tfpf, 2024-11-11:
The AUR package `python-control-git` is out of date. It was last updated in 2022. See https://aur.archlinux.org/packages/python-control-git. At that point in time, this repository presumably had a `setup.py`.

There’s also `python-control`: https://aur.archlinux.org/packages/python-control. It’s also out of date, unfortunately. 

Is there a good reason to have a system-wide installation? Can you try a virtual environment with access to system site packages?

vibhoregit, 2024-11-11:
I was hoping that system wide installation would save me creating venv(s), each with its own version of 'latest' packages and not having to worry about updates as pacman would do it for me. But if package management requires me to create a venv to ensure no broken dependencies, then so be it. Thanks :)

closed 2024-11-11 by vibhoregit