python-control/python-control #1095: Nyquist plot indent direction doesn't seem to work

JonHowMIT, 2025-01-14:
Hi - When I run this code in Jupyter notebook I get different results:

from platform import python_version
print("Running Python:",python_version())
! pip list | grep control
from control import nyquist_plot, zpk
import matplotlib.pyplot as plt

G = zpk([], [0, -2, -3], gain=100)

fig, ax = plt.subplots(1,figsize=(4, 4),dpi=150,constrained_layout = True)
out = nyquist_plot(G,indent_points=100,indent_direction='left',indent_radius=0.25)

fig, ax = plt.subplots(1,figsize=(4, 4),dpi=150,constrained_layout = True)
out = nyquist_plot(G)

1) Running Python: 3.8.19 and control  0.9.4 it works great and get pretty plot
2) Running Python: 3.12.6 and control 0.10.1 I get: "AttributeError: Line2D.set() got an unexpected keyword argument 'indent_points' "

any thoughts?
thx
jon

p.s. would really like to understand better how these indent plot are made, bec the first "pretty plot" in (1) above is really hard to intepret

JonHowMIT, 2025-01-14:
and if I remove the indent_points, then the error is:

AttributeError: Line2D.set() got an unexpected keyword argument 'indent_direction'

JonHowMIT, 2025-01-14:
oh, and the reference text (possibly out of date) for 0.10.1 says these are options:

indent_direction (str, optional) – For poles on the imaginary axis, set the direction of indentation to be ‘right’ (default), ‘left’, or ‘none’.

indent_points (int, optional) – Number of points to insert in the Nyquist contour around poles that are at or near the imaginary axis.

indent_radius (float, optional) – Amount to indent the Nyquist contour around poles on or near the imaginary axis. Portions of the Nyquist plot corresponding to indented portions of the contour are plotted using a different line style.

murrayrm, 2025-01-14:
Something is messed up in the way that arguments are being processed. I think it works OK if you split into two commands: computing the Nyquist response and plotting it:
```
import control as ct
import matplotlib.pyplot as plt

G = ct.zpk([], [0, -2, -3], gain=100)
resp = ct.nyquist_response(G,indent_points=100,indent_direction='left',indent_radius=0.25)
resp.plot()
```
[image]

JonHowMIT, 2025-01-14:
yep that works - is "nyquist_response" new since 0.9.4 as it doesn't work in that version (hard to know when things are added)

closed 2025-01-16 by murrayrm