python-control/python-control #1101: pyflakes (pylint, ruff, ....) clean-up, CI?

roryyorke, 2025-01-19:
Running pyflakes on main HEAD (0ff045263f) reveals lots of possible issues; see end.

1. Is there interest in getting and keeping the tree lint-clean?

    This would require some changes to style, e.g., dropping `from foo import *` (see output below), and some patience when the linter spits out an incomprehensible message.
 
2. Are there strong preferences for a particular linter?

    I use pyflakes often: it's simple, has a low false-positive rate, but is not configurable.

    The other two I'm aware of:
      - pylint, more extensive checks, configurable
      - ruff; my crude, probably unfair thinking of it is "pylint written in Rust", but I think it may do more

pyflakes output:

```
control/modelsimp.py:343:9: local variable 'order' is assigned to but never used
control/pzmap.py:14:1: 'math.pi' imported but unused
control/pzmap.py:18:1: 'numpy.cos' imported but unused
control/pzmap.py:18:1: 'numpy.exp' imported but unused
control/pzmap.py:18:1: 'numpy.linspace' imported but unused
control/pzmap.py:18:1: 'numpy.sin' imported but unused
control/pzmap.py:18:1: 'numpy.sqrt' imported but unused
control/pzmap.py:25:1: '.freqplot._freqplot_defaults' imported but unused
control/pzmap.py:28:1: '.lti.LTI' imported but unused
control/stochsys.py:25:1: '.iosys.InputOutputSystem' imported but unused
control/stochsys.py:25:1: '.iosys._process_indices' imported but unused
control/stochsys.py:29:1: redefinition of unused 'StateSpace' from line 23
control/stochsys.py:173:9: local variable 'NN' is assigned to but never used
control/stochsys.py:292:9: local variable 'NN' is assigned to but never used
control/stochsys.py:449:13: undefined name 'warnings'
control/stochsys.py:463:9: 'raise NotImplemented' should be 'raise NotImplementedError'
control/iosys.py:11:1: 'warnings.warn' imported but unused
control/nlsys.py:21:1: 'copy' imported but unused
control/nlsys.py:28:1: '.iosys._process_signal_list' imported but unused
control/nlsys.py:214:9: local variable 'dt' is assigned to but never used
control/nlsys.py:246:9: local variable 'dt' is assigned to but never used
control/nlsys.py:814:16: f-string is missing placeholders
control/nlsys.py:834:16: f-string is missing placeholders
control/nlsys.py:2477:5: '.statesp._convert_to_statespace' imported but unused
control/nlsys.py:2478:5: '.xferfcn.TransferFunction' imported but unused
control/nlsys.py:2540:12: f-string is missing placeholders
control/nlsys.py:2579:16: f-string is missing placeholders
control/nlsys.py:2642:24: f-string is missing placeholders
control/nlsys.py:2668:16: f-string is missing placeholders
control/nlsys.py:2745:24: f-string is missing placeholders
control/mateqn.py:40:1: 'numpy.copy' imported but unused
control/mateqn.py:40:1: 'numpy.dot' imported but unused
control/mateqn.py:40:1: 'numpy.atleast_2d' imported but unused
control/statesp.py:19:1: 'copy.deepcopy' imported but unused
control/statesp.py:25:1: 'numpy.array' imported but unused
control/statesp.py:25:1: 'numpy.ones' imported but unused
control/statesp.py:35:1: '.iosys._process_dt_keyword' imported but unused
control/statesp.py:35:1: '.iosys.iosys_repr' imported but unused
control/statesp.py:35:1: '.iosys.isdtime' imported but unused
control/nichols.py:24:1: '.freqplot._default_frequency_range' imported but unused
control/nichols.py:24:1: '.freqplot._freqplot_defaults' imported but unused
control/nichols.py:140:29: undefined name 'idx_sys'
control/phaseplot.py:41:1: '.exception.ControlNotImplemented' imported but unused
control/phaseplot.py:165:5: local variable 'passed_kwargs' is assigned to but never used
control/phaseplot.py:633:14: local variable 'single_color' is assigned to but never used
control/phaseplot.py:1094:13: f-string is missing placeholders
control/phaseplot.py:1096:19: undefined name 'ControlArgument'
control/phaseplot.py:1096:35: f-string is missing placeholders
control/phaseplot.py:1261:9: local variable 'xy' is assigned to but never used
control/phaseplot.py:1268:9: local variable 'bp' is assigned to but never used
control/statefbk.py:49:1: '.exception.ControlNotImplemented' imported but unused
control/statefbk.py:53:1: '.mateqn._check_shape' imported but unused
control/statefbk.py:55:1: '.statesp._convert_to_statespace' imported but unused
[... 244 lines cut ...]
control/matlab/__init__.py:86:1: '..stochsys.dlqe' imported but unused
control/matlab/__init__.py:90:14: 'poles' may be undefined, or defined from star imports: ..bdalg, ..ctrlutil, ..delay, ..dtime, ..frdata, ..iosys, ..lti, ..mateqn, ..modelsimp, ..pzmap, ..statefbk, ..statesp, ..xferfcn
control/matlab/__init__.py:90:21: 'zeros' may be undefined, or defined from star imports: ..bdalg, ..ctrlutil, ..delay, ..dtime, ..frdata, ..iosys, ..lti, ..mateqn, ..modelsimp, ..pzmap, ..statefbk, ..statesp, ..xferfcn
control/matlab/__init__.py:91:12: 'frequency_response' may be undefined, or defined from star imports: ..bdalg, ..ctrlutil, ..delay, ..dtime, ..frdata, ..iosys, ..lti, ..mateqn, ..modelsimp, ..pzmap, ..statefbk, ..statesp, ..xferfcn
control/matlab/__init__.py:95:1: 'from .timeresp import *' used; unable to detect undefined names
control/matlab/__init__.py:96:1: 'from .wrappers import *' used; unable to detect undefined names
control/matlab/__init__.py:99:1: 'from ..config import *' used; unable to detect undefined names
control/matlab/__init__.py:100:1: 'use_matlab_defaults' may be undefined, or defined from star imports: ..bdalg, ..config, ..ctrlutil, ..delay, ..dtime, ..frdata, ..iosys, ..lti, ..mateqn, ..modelsimp, ..pzmap, ..statefbk, ..statesp, ..xferfcn, .timeresp, .wrappers
control/flatsys/bspline.py:11:1: 'scipy.interpolate.splev' imported but unused
control/flatsys/__init__.py:64:1: '.basis.BasisFamily' imported but unused
control/flatsys/__init__.py:65:1: '.poly.PolyFamily' imported but unused
control/flatsys/__init__.py:66:1: '.bezier.BezierFamily' imported but unused
control/flatsys/__init__.py:67:1: '.bspline.BSplineFamily' imported but unused
control/flatsys/__init__.py:70:1: '.systraj.SystemTrajectory' imported but unused
control/flatsys/__init__.py:71:1: '.flatsys.FlatSystem' imported but unused
control/flatsys/__init__.py:71:1: '.flatsys.flatsys' imported but unused
control/flatsys/__init__.py:72:1: '.linflat.LinearFlatSystem' imported but unused
control/flatsys/__init__.py:75:1: '.flatsys.point_to_point' imported but unused
control/flatsys/__init__.py:75:1: '.flatsys.solve_flat_ocp' imported but unused
control/flatsys/flatsys.py:248:5: '..iosys._process_iosys_keywords' imported but unused
control/flatsys/flatsys.py:486:19: undefined name 'ControlArgument'
control/flatsys/flatsys.py:520:13: redefinition of unused 'traj_cost' from line 499
control/flatsys/flatsys.py:724:5: local variable 'Tf' is assigned to but never used
control/flatsys/flatsys.py:725:46: undefined name 'T0'
```

roryyorke, 2025-01-19:
See https://github.com/python-control/Slycot/pull/246 for applying `ruff check` to the much smaller Slycot Python source.

See [here](https://github.com/roryyorke/Slycot/actions/runs/12852452949/job/35834440221) for CI test result failing with `ruff check`.

murrayrm, 2025-01-19:
> Is there interest in getting and keeping the tree lint-clean?

Yes!

> Are there strong preferences for a particular linter?

Not on my end..  I've tried running `ruff` before and it generated a lot of warnings about fairly minor things.  The `pyflake` output seems about right (all of those items seem like things we should fix).

slivingston, 2025-01-20:
Yes, let's do this!

In the past I used `pylint` regularly, but I switched to `ruff` because it is much faster. `ruff` (and `pylint`) will require significant tuning effort to align with python-control style, so I recommend the concrete goal of requiring `pyflakes` to pass in CI jobs. After we try that in practice, we can consider tightening it to `ruff`, `pylint` etc.

roryyorke, 2025-02-01:
after struggling with pyflakes, I'm  try ruff with "pyflakes-only" warnings, and add some "# noqa" lines to `__init__.py`.

murrayrm, 2025-02-03:
Addressed in #1118.  OK to close, unless there is additional discussion of what we want to do.