python-control/python-control #1119: Sporadically failing CI tests

murrayrm, 2025-02-02:
In the last few months, it seems like the CI tests are failing much more often.  Re-running them usually fixes the problem, but perhaps there is something we are doing that exacerbates the failures?

See some comments [here](https://github.com/python-control/python-control/pull/1118#issuecomment-2629352703).

slivingston, 2025-02-17:
This may be a problem with cache management. A similar error is reported at https://github.com/conda-forge/miniforge/issues/513 and the recommended solution is to clear the cache (`conda clean`), but doing so in CI jobs may incur a large (aggregate) time and bandwidth cost.

murrayrm, 2025-02-17:
Thanks for looking into this, @slivingston.  It is not that huge a pain to re-run the tests and I think that is probably preferable to imposing a large time/bandwidth cost for every push.