python-control/python-control #1131: root locus plot omega-damping grid appearing by default

mjdunlop, 2025-02-26:
I'm encountering a minor issue with `control.root_locus`. The documentation says the default behavior has the omega-damping grid set to False, but it's appearing unless I actually set `grid = False`.

Here are the versions I'm working with:
control 0.10.1
matplotlib 3.10.0
I'm seeing this in both a Jupyter Notebook and Google Colab.

Here's an example of the issue:
[image]
[image]

Here's what it looks like if `grid = False`:
[image]

slivingston, 2025-02-27:
Can you point to where in the documentation the default value is indicated?

There is a new release coming soon (https://github.com/python-control/python-control/issues/1130) that might already include a fix for this, but [the function description in the current release docs](https://python-control.readthedocs.io/en/0.10.1/generated/control.root_locus.html#control-root-locus) seem to have the correct default, i.e., [the value of config.defaults['rlocus.grid']
](https://github.com/python-control/python-control/blob/f6799ab8e60d7fcd4b02d52d5903f08c24957faf/control/rlocus.py#L30-L32)

mjdunlop, 2025-02-27:
Ah! I think my google searching had turned up a previous version of the documentation ([here](https://python-control.readthedocs.io/en/0.9.4/generated/control.root_locus.html)). So it looks like the default behavior actually is okay. Thanks!

closed 2025-02-27 by mjdunlop

slivingston, 2025-02-28:
[ReadTheDocs will add a warning banner](https://docs.readthedocs.com/platform/stable/versions.html#version-warning-notifications) when viewing docs from older versions or from current development (`main` branch, but not released yet) if a `stable` version is known. For example, see the warning on the mypy "latest" docs: <https://mypy.readthedocs.io/en/latest/builtin_types.html>

<https://python-control.readthedocs.io> does not show a "stable" option, so I guess that ReadTheDocs is failing to detect that 0.10.1 (the most recent version-like tag) should be "stable." Until we fix this, a warning banner will not be shown.

@murrayrm can you access the ReadTheDocs administrative view to check whether there is any indication of why "stable" is not being labeled? Feel free to delegate...I do not have ReadTheDocs admin access and don't know who does.