python-control/python-control #1132: Latex is not rendered in HTML output of in VSCode

lkies, 2025-03-01:
On f6799ab8e60d7fcd4b02d52d5903f08c24957faf

As the title says, the math sections in HTML output generated by `_repr_html_` from `StateSpace` (and other classes) is not rendered.

[image]

This is not really the fault of this library and there is an open [issue](https://github.com/microsoft/vscode/issues/203725) in the VSCode repo for this, but the simpler output which used `_html_latex_` before #1091 did not have this problem.

[image]

I think adding back the old `_repr_latex_` would allow the user to choose between the new mixed HMTL output and the old Latex output if using VSCode, what do you think?

Another option that I saw in the VSCode issue is to explicitly add the renderer into the HTML but this seems really hacky:

[image]

murrayrm, 2025-03-01:
A couple of possibilities:
* If have `_repr_html` and `_repr_latex` allows VSCode to use LaTeX and Jupyter to use HTML, that could work.
* We could add something into `_repr_html` that checks for VSCode and set the output accordingly (won't be as pretty, but should work OK).
* We could add an option that allows the default `repr` format to be changed and users could use then set the format to `vscode` (or something like that), which would give the older style format.
I did try at one point getting things to look good with just LaTeX, but it was very hard to convince the LaTeX processor in Jupyter to do the right thing (it wanted raw latex math code, if memory serves).

lkies, 2025-03-01:
I experimented a bit more and it seems that VSCode will render latex contained in `_repr_markdown_` which mostly behaves like a superset of `_repr_html_`. So one can simply switch the output to text/markdown and everything works pretty much as it should have in the first place.

Pull request for blanked `_repr_markdown_` comming in a few minutes.

PR #1133 (2025-03-03): Fix Latex not being rendered in HTML output in VSCode
...by adding a blanket `_repr_markdown_` for `InputOutputSystem` which is the same as `_repr_html_` but the renderer for `_repr_markdown_` will also render contained Latex.

Closes #1132
closed 2025-03-03 by murrayrm