python-control/python-control #1141: Casting complex number to float

Master-Pr0grammer, 2025-03-21:
Code to reproduce the problem:
```python3
sys = ctrl.zpk([], [1j], 1)

plt.figure(figsize=(8, 6))
ctrl.root_locus(sys, color='red') #<-- Problem is here

plt.xlabel('Real Axis (σ)')
plt.ylabel('Imaginary Axis (jω)')
plt.grid(True)
plt.show()
```

In the diagram, you can see the pole is placed at the wrong location. but I looked into it and found a quick solution (though I am not familiar with the code, and it might be deeper than I found)

in the files xferfcn.py (lines 1068, and 1072) there are two arrays that are created of type `float`, that could later assigned to complex numbers if the input is of type `complex`. the fix is to just change the type of the array to `complex`

Then another issue arises in plocus.py (line 267), if the coefficients are complex, it is again casted into a float. I am not sure what this value is used for, but I believe this can be fixed by casting the input type to complex if it isn't already, then using the magnitude of the complex number instead of casting it, which I believe the abs() function will do for default python `complex` types.

Should be an easy fix, hope this helps

roryyorke, 2025-03-23:
Thanks for the report.  Supporting complex coefficients for TFs, and equivalently complex matrices for SS, is brought up now and then.  It's unfortunately not such an easy change: see gh-371.

Master-Pr0grammer, 2025-03-23:
ah I see now, Might be worth looking into adding type checking to throw errors for such a case since the `complex` type is a default type in python, and the client is likely going to try to use it.

closed 2025-03-25 by bnavigator