python-control/python-control #1147: too much white space around Root Locus with omega-damping grid

datacurse, 2025-04-26:
When I turn on the omega-damping grid for root locus plot with `grid=True` (turned on by default), it adds too much white space around the actual plot, which is very annoying. With this grid turned off, plot is tight and nice.

I would like root locus to not have this extra white space around when grid is turned on, I am pretty sure this is some sort of platform-specific bug. Would appreciate any help 🙏

grid turned on (probably bug):
[image]

grid turned off (expected behavior):
[image]

Libraries:
- control 0.10.1
- matplotlib 3.10.0

I'm using python control in Google Colab.

slivingston, 2025-04-26:
I cannot reproduce on the current tip of `main` branch, so this must have been fixed already. There is a new release coming soon: https://github.com/python-control/python-control/issues/1130

murrayrm, 2025-04-26:
@slivingston Did you run inside a Jupyter notebook?  What version?

I tried running this using the following versions and got the same problem as @datacurse noted:
```
IPython          : 8.31.0
ipykernel        : 6.29.5
ipywidgets       : 8.1.5
jupyter_client   : 8.6.3
jupyter_core     : 5.7.2
jupyter_server   : 2.14.2
jupyterlab       : 4.3.3
nbclient         : 0.10.1
nbconvert        : 7.16.4
nbformat         : 5.10.4
notebook         : 7.3.1
qtconsole        : not installed
traitlets        : 5.14.3
```

I also tried running this on Colab and I get the same problem using the following commands:
```
!pip install git+https://github.com/python-control/python-control.git

import control as ct
import matplotlib.pyplot as plt

sys = ct.tf([1, 2], [1, 0, 0])
cplt = ct.root_locus(sys, grid=True)
```

slivingston, 2025-04-26:
No, I ran it in IPython in a terminal, so the Matplotlib backend would be different. I just tried in Colab and reproduced it (like you), so now I think it is something new.