python-control/python-control #1156: provide more prominent interface for lft linear fractional transformation

ovhnw, 2025-06-22:
so it's easier to find or know about. or explain in the [function reference](https://python-control.readthedocs.io/en/0.10.1/control.html) that there are other more functions available through state space instances. so you at least know where else to look for capabilities.

murrayrm, 2025-06-22:
If someone wants to take this on, there are a couple of other enhancements that it would be good to address at the same time:

* The current docstring for `StateSpace.lft` doesn't provide any information on what the function does, it just references a textbook and (tries to) link to the MATLAB documentation (the link is broken).  It would be better to have a self-contained docstring.
* The current method does not preserve signal names for the inputs and outputs of the transformed system.
* The current method does not allow renaming of the system, inputs, outputs, and states.  This should be done in the standard way of all system creation functions.
* It might be useful to create an `lft` function in the `control` namespace with signature `lft(sys1, sys2, [nu, ny], **kwargs)`.  If `sys1` and `sys2` are `StateSpace` objects, the current method could be used.  Otherwise, the `interconnect` function could be called to create the system for any I/O system.
* Narrative documentation should be placed in the User Guide, either in Chapter 3 (Linear Systems) or Chapter 6 (Interconnected I/O Systems).  The latter probably makes the most sense, perhaps in the Block Diagram Algebra section.
* Function documentation should be placed in the Reference Manual, likely in the System Interconnection section of the Function Reference chapter.

Note: References to sections in the documentation are relatively to the current main branch, which was reorganized in #1094.

murrayrm, 2025-06-24:
`singular_values_plot` is in the function reference for the current main branch (which will become 0.10.2):

https://python-control.readthedocs.io/en/latest/functions.html#control-system-analysis