python-control/python-control #1158: silent failure to find mimo zeros for state space

ovhnw, 2025-06-23:
```
from control import tf, zeros, ss

sys = tf([[    [1, 1],  [1]  ],
            [     [1],     [1]     ]],
            [[    [1, 5],  [1]  ],
            [     [1],     [1]     ]])
            
print(zeros(sys))
```
results in `NotImplementedError: TransferFunction.zeros is currently only implemented for SISO systems.`. However, converting the system to state space fails silently with no warning. For example:

```
from control import tf, zeros, ss

sys = ss(tf([[    [1, 1],  [1]  ],
            [     [1],     [1]     ]],
            [[    [1, 5],  [1]  ],
            [     [1],     [1]     ]]))
            
print(zeros(sys))

```
returns `[]`

murrayrm, 2025-07-04:
I took a look at this and I think the answer is correct.  The MIMO transfer function that you have created is
```
[  s + 1          ]
[  -----      1   ]
[  s + 5          ]
[                 ]
[    1        1   ]
````
While this has a zero in the (1, 1) entry of the matrix transfer function, as a *MIMO* system this is not a transmission zero since it is not a zero in the other channels (specifically the (2, 1) entry).

Here's a different example that shows a MIMO zero:
```
sys_tf = ct.tf(
    [[    [1, 1],  [1]  ], [     [1, 1],     [1]     ]],
    [[    [1, 5],  [1]  ], [    [1, 2],     [1]     ]])

sys_ss = ct.ss(sys_tf)
```
This generates the transfer function
```
[  s + 1          ]
[  -----      1   ]
[  s + 5          ]
[                 ]
[  s + 1          ]
[  -----      1   ]
[  s + 2          ]
```
and we now have transmission zero in the first input channel at $s = -1$. 

If we compute the zeros using python-control, we get:
```
In [4]: sys_ss.zeros()
Out[4]: array([-1.+0.j])
```

closed 2025-07-04 by murrayrm

ovhnw, 2025-07-04:
Ok, honestly i didn't know how zeros were defined for MIMO systems. It just seemed off that it should throw an error in a state space representation, but not a transfer function representation. :D thanks for teaching me something.

Maybe change the error message to `NotImplementedError: TransferFunction.zeros is currently only implemented for SISO systems. As a workaround convert your system to StateSpace.` That would clarify that MIMO systems are supported, just not while in the TransferFunction representation.