python-control/python-control #1159: lqe does not work with 3 arguments

ovhnw, 2025-06-23:
```
from control import lqe,ss,ctrb,obsv
from numpy import diag,array
from numpy.linalg import matrix_rank

A = array([ [0,1,0],
            [0,0,1],
            [0,0,0]])
B = array([[0],[0],[1]])
C = array([[1,0,0]])
D = array([[0]])

sys=ss(A,B,C,D)
print(matrix_rank(ctrb(A,B)))
print(matrix_rank(obsv(A,C)))

L,_,_=lqe(A,diag([1,1,1]),C,diag([1,1,1]),array([[1]]))
print(L)
L,_,_=lqe(sys,diag([1,1,1]),array([[1]]))

```

outputs

```
3
3
[[2.41421356]
 [2.41421356]
 [1.        ]]
Traceback (most recent call last):
  File "/home/paul/Documents/paul/personal/properly named/tippy control with better control/test_lqe.py", line 19, in <module>
    L,_,_=lqe(sys,diag([1,1,1]),array([[1]]))
          ~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/paul/Documents/paul/personal/properly named/tippy control with better control/.venv/lib/python3.13/site-packages/control/stochsys.py", line 176, in lqe
    _check_shape("QN", QN, G.shape[1], G.shape[1])
    ~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/paul/Documents/paul/personal/properly named/tippy control with better control/.venv/lib/python3.13/site-packages/control/mateqn.py", line 643, in _check_shape
    raise ControlDimension("Incompatible dimensions of %s matrix" % name)
control.exception.ControlDimension: Incompatible dimensions of QN matrix

```
if the G matrix is manually specified `lqe` works, otherwise it seems to find a dimension mismatch

sawyerbfuller, 2025-07-04:
I remember when we chose to use B as G when passed a system, and that I wasn’t convinced one way or the other at the time whether that was the right choice. 

It may be (I will have to double check) that Octave is assuming G=diag(ones(n)), which may turn out to be a better assumption. 

What do you think?

ovhnw, 2025-07-04:
If you don't mind my thoughts:

- My college course notes "book" does not use a G matrix. (implied identity?)
- There is a duality between LQR and LQE. They have the same Algebraic Riccati Equation structure. So you can do `lqr(A,C',Q,R)'` (or something, i don't remember for sure offhand) as an equivalent for lqe. If there is no dual to the G matrix in lqr (is there? i'm not sure), then perhaps the identity may be the only default that preserves this duality.
- I took a quick look at matlab's `kalman` and G seems to be _included as part of_ B.

My vote would go to an identity for default. 

>The one thing we do need to take care of is documenting the fact that if G is not specified, then its default value is B .

:D yes. and thank you.

murrayrm, 2025-07-04:
Thoughts definitely welcome @ovhnw!

I don't have a strong preference, as long as we leave in the option for $G$ to be specified somehow.  We should also be aware that if we change the default, we could break existing code => we should probably either issues a warning or provide some other indication that behavior has changed (and we might wait until v0.11.x before changing the default).

Reopening this issue, but changing the title.

ovhnw, 2025-07-04:
i should point out, although the current behavior deceptively differs from octave's `lqe`, it may be compatible with matlab's `kalman`. You might even be able to adapt your code to accept the `[B G]` style B matrices without breaking existing code.

sidenote: it is unfortunate that matlab decided to format the B matrices as `[B G]`, instead of `[G B]`, which is the convention for robust control.