python-control/python-control #1162: can't pickle serialize state space objects

ovhnw, 2025-06-28:
trying to pickle a state space object causes error

```
from control import ss
from pickle import dump

sys=ss([[0,1],[0,0]],[[0],[1]],[[1,0]],[[0]])

with open("deleteme","wb") as f:
    dump(sys,f)

```

```
Traceback (most recent call last):
  File "/home/paul/Documents/paul/personal/properly named/tippy control with better control/test_pickle_ss.py", line 7, in <module>
    dump(sys,f)
    ~~~~^^^^^^^
AttributeError: Can't get local object 'StateSpace.__init__.<locals>.<lambda>'

```
control         0.10.1
Python 3.13.3
archlinux

slivingston, 2025-06-29:
The implementation of state space objects has features that make unable to be pickled with [`pickle`](https://docs.python.org/3/library/pickle.html). However, you can pickle/unpickle it using [dill](https://github.com/uqfoundation/dill):

```python
from control import ss
from dill import dump, load

sys=ss([[0,1],[0,0]],[[0],[1]],[[1,0]],[[0]])

with open("deleteme","wb") as f:
    dump(sys,f)

with open("deleteme", "rb") as f:
    syscopy = load(f)

print(syscopy)
```

slivingston, 2025-06-29:
I have not tested with `dill` much , but dill is more than 15 years old and

closed 2025-06-29 by slivingston

ovhnw, 2025-06-29:
thanks for the workaround