python-control/python-control #1168: model reduction by residualization not implemented or not documented

ovhnw, 2025-07-13:
the documentation does not explain what exactly "matchdc" does. is it residualization or something else? It seems to not just be a gain adjustment.

murrayrm, 2025-07-13:
From the `model_reduction` docstring:
```
 Two methods of state reduction are possible: 'truncate' removes the
 states marked for elimination, while 'matchdc' replaces the eliminated
 states with their equilibrium values (thereby keeping the input/output
 gain unchanged at zero frequency ["DC"]).
```

ovhnw, 2025-07-13:
> From the model_reduction docstring: ...

ok that smells like the residualization described by skogestad, but i can't tell for sure

My bad again. I was reading docs [here](https://python-control.readthedocs.io/en/0.10.1/generated/control.model_reduction.html#control.model_reduction), which do not contain as descriptive documentation. It must have improved since then. Searching for the docstring you quoted i found that there is a "latest" url for readthedocs.

murrayrm, 2025-07-13:
You should get a warning if you reference an older version of the documentation (see [here](https://github.com/python-control/python-control/pull/1140#issuecomment-2726755594)).

An open issue for anyone reading this and having some time to contribute: while 'matchdc' is described in `model_reduction`, it is not described in `balanced reduction`, where it is also used.

ovhnw, 2025-07-13:
> You should get a warning if you reference an older version of the documentation (see https://github.com/python-control/python-control/pull/1140#issuecomment-2726755594).

sure enough - i refreshed the page and there is a tooltip that shows up for 4 seconds when the page is refreshed. sounds like a readthedocs problem, but it should probably be more prominent.

murrayrm, 2025-07-20:
FYI, I've got a branch of small documentation changes and I have updated the documentation for `balanced_reduction` there, so this will get addressed for 0.10.3.