python-control/python-control #1180: Initial conditions are normalised by coefficient of s^n in denominator

vibhoregit, 2025-09-04:
Given minimum working example below

'num = [1]
den   = [3,5,26]
sys = ct.tf(num,den)
X0 = [5,3]
T, yout = ct.initial_response(sys,X0 = [D,M])
plt.plot(T, yout,color='red')
plt.show()
'

As per initial conditions, output x(0) = 3 but in plot of inital response x(0) = 1. To get correct initial condition in plot, X0 = [15,9]. The code is somehow normalizing the initial condition with den[0] coefficient

murrayrm, 2025-09-05:
The state space system associated with a transfer function is not unique and so specifying the initial condition is not well-defined.

Taking the example that you gave, you can see the state space representation that python-control uses if you convert to state space explicitly:
```
sys_tf = ct.tf(num,den)
sys_ss = ct.ss(sys_tf)
print(sys_ss)
```
This gives:
```
Inputs (1): ['u[0]']
Outputs (1): ['y[0]']
States (2): ['x[0]', 'x[1]']

A = [[-1.66666667 -8.66666667]
     [ 1.          0.        ]]

B = [[-1.]
     [ 0.]]

C = [[ 0.         -0.33333333]]

D = [[0.]]
```
We see from this that if we set the initial condition to [5,3], then the initial output will be -1.

If you wanted to have the initial output be 1, you could use the following alternative realization:
```
sys_alt = ct.ss(sys_ss.A, sys_ss.B / -3, sys_ss.C * -3, 0)
print(ss.tf(sys_alt))
```
This has the same transfer function
```
<TransferFunction>: sys[9]
Inputs (1): ['u[0]']
Outputs (1): ['y[0]']

         0.3333
  ---------------------
  s^2 + 1.667 s + 8.667
```
but the initial output if you set X0=[5, 3] would now be 3.

closed 2025-09-05 by murrayrm

vibhoregit, 2025-09-05:
Thank you for the explanation, I can write a wrapper function for my case now!